chariotsolutions/phonegap-nfc #283: nfc.beginSession not opening the dialogue over the app

andyj, 2017-09-27:
I have a button in my app which is simply calling the following function 

```
function iosNFC(){
  nfc.beginSession(
    function(){console.log("Success");},
    function(){console.log("ERROR");}
  );
}
```

The problem is the "READY TO SCAN" box doesn't pop up over the app. The only way I know its there is if I double click the home button to switch away and then it takes focus.

Is this an iOS issue or app problem?

scottire, 2017-09-27:
Have you turned on the Near Field Communication Tag Reading capability in your Xcode project?

don, 2017-10-03:
@andyj are you still seeing this? What version of iOS are you using?

Can you share a project on GitHub so I can try to duplicate on my hardware?

andyj, 2017-10-03:
Version 11.0.1 (15A403)
iPhone 8+ ( if that changes anything?)

don, 2017-10-03:
I was hoping 11.0.1 might have fixed the issue. Can you share a simple project? I haven't been able to duplicate this.

andyj, 2017-10-03:
I've just created https://github.com/andyj/iosnfc

JohnMcLear, 2017-10-03:
I don't experience this bug, weird one..   If you create the project with cordova do you get the same experience?

I tested on iPhone 7 & 8.  Not sure on version.

andyj, 2017-10-03:
I’ll try Cordova tonight. My only thought is that an Xcode setting is affecting it?

andyj, 2017-10-03:
@JohnMcLear thank you for pointing me at Cordova. Works fine. The problem then is Phonegap, my installed version is up to date with [the current releases](https://build.phonegap.com/current-support) too.

JohnMcLear, 2017-10-03:
Pleasure.

closed 2017-10-04 by andyj

JoshuaPerk, 2017-12-24:
@andyj I seem to be having an issue that appears to be rooted in the same vein. I took your repo and added platform ios in cordova, then built, and ran it on my device via xcode. I hit the button and nothing seems to happen (other than the "NFC button hit" text). I had to add a few Capabilities in order to compile, but I'm not quite sure what I'm missing... (iPhone 7).

JoshuaPerk, 2017-12-24:
@andyj I was in-fact able to get it sort-of working. Running into the same issue though. Doesn't pop up until I double click the home button... No idea why...

JoshuaPerk, 2017-12-29:
@don @JohnMcLear, are you aware of any xCode settings or quarks with NFC Core or the plugin implementation that would be causing the following? I've rebuilt in the most recent Cordova and PhoneGap to no avail.

[](https://youtu.be/LwilT02EFxg)

@andyj, I'm assuming the above is what you experienced?

JohnMcLear, 2017-12-31:
I don't, sorry!  I'd offer to try build for you but I'm not at my iOS build environment for another month or so.

JoshuaPerk, 2017-12-31:
I greatly appreciate the notion; no worries! I believe the issue was (oddly enough) tied directly to the CSP tag. My CSP settings were preventing the dialogue modal from appearing. When I exited the app (and my CSP rules were no longer in place), the modal executed.

Happy New Year!