chariotsolutions/phonegap-nfc #286: UID on iOS

erikm30, 2017-10-03:
According to https://github.com/hansemannn/iOS11-NFC-Example/issues/16 it should be possible to access UID and tag type also on iOS.
Would it be possible to adapt the tag object of the NFCEvent to show the id as well ?

don, 2017-10-03:
If Apple added APIs for UID and TagType, it can be added to the plugin.

erikm30, 2017-10-04:
There is no direct API, but according to the link above it is possible to access the data.

scottire, 2017-10-04:
I created a pull request https://github.com/chariotsolutions/phonegap-nfc/pull/287

The interface is not public and is only gotten through `valueForKey` accessors on private instance variables which is not ideal. They should be exposed from Apple but this should do until then

don, 2017-10-27:
@scottire I'm going to close the issue since apple rejected an app that used this (See https://github.com/chariotsolutions/phonegap-nfc/pull/287#issuecomment-340043226)

closed 2017-10-27 by don