chariotsolutions/phonegap-nfc #289: Is this normal?

rolandjitsu, 2017-10-10:
I'm trying to debug my app and why it crashes and I've noticed that this plugin logs the following:
```
10-10 13:52:41.875  4795  4844 D NfcPlugin: execute enabled
10-10 13:52:41.876  4795  4844 D NfcPlugin: execute enabled
...
```

It keeps logging this forever, until the app crashes. I suspect that this should not normally happen.

I'm using this plugin with Ionic and [NFC](https://ionicframework.com/docs/native/nfc/) plugin wrapper.

My current Ionic env:
```
cli packages: (/Users/...)

    @ionic/cli-utils  : 1.12.0
    ionic (Ionic CLI) : 3.12.0

global packages:

    cordova (Cordova CLI) : 7.0.1 

local packages:

    @ionic/app-scripts : 3.0.0-201710101530
    Cordova Platforms  : none
    Ionic Framework    : ionic-angular 3.7.1

System:

    Android SDK Tools : 25.2.5
    Node              : v8.6.0
    npm               : 2.15.12 
    OS                : macOS Sierra
    Xcode             : Xcode 9.0 Build version 9A235 
```

don, 2017-10-23:
NFC code won't work in emulators. It needs a real NFC chip.

rolandjitsu, 2017-10-23:
@don I see. But perhaps it would be a good idea not to crash the app, it makes it impossible to debug in the emulator otherwise.

But this could be something that I'm doing wrong ...

don, 2017-10-23:
I agree that crashing is not good. In the past nfc stuff just didn’t work. Maybe it’s an issue in the Ionic native wrapper or different plugin behavior with Gennymotion. 

Can you duplicate this in a simple Cordova app without Ionic?

rolandjitsu, 2017-10-24:
It's possible, there has been a lot of changes. It might have been fixed just recently, but let me test it again and I'll get back to you.

closed 2018-04-18 by don