chariotsolutions/phonegap-nfc #295: Nfc plugin with ionic doesnt work on some devices

Vartex05, 2017-10-23:
Iam using NFC in my app for reading personal chips. Problem is, that exactly same app, with exactly same code, works differently on differenct devices. I tested it on Samsung A5,A3 and Nokia 3. NFC works on A5 and Nokia 3, but on A3, Nfc doesnt work. It reads the chip (because it shows android system popup, that this nfc tag is not supported), but no event in my app is raised. Iam using addTagDiscoveredListener method for listening for events. Any ideas why this happens, and how to fix it?

don, 2017-10-23:
> it shows android system popup, that this nfc tag is not supported

There's nothing I can do if your phone doesn't support the tags. Generally phones with Broadcom NFC chips won't support Mifare Classic tags. Try using a different NFC tag type.

closed 2017-10-23 by don

Vartex05, 2017-10-23:
The problem is that it is not Mifare classic tag, its proximity chip. On A5 or Nokia3 it also shows message that this tag is not supported, but it reads it...