chariotsolutions/phonegap-nfc #296: Andorid example don't work

raxon, 2017-10-23:
When i use this https://github.com/chariotsolutions/phonegap-nfc/blob/master/doc/GettingStartedCLI.md
example and compile, install it dosn't work.

I compiled, installed and got information that put nfc into the device but when I put it nothing happens

don, 2017-10-23:
What phone are you using? What version of Android? Is NFC enabled?

Are there any errors from `adb logcat` or javascript errors using chrome://inspect?

What type of NFC tag are you using? Is there an NDEF message on the tag?

Try reading the tag with https://play.google.com/store/apps/details?id=com.nxp.nfc.tagwriter. If that works write a new Text message back to the tag and try reading again with your app.

raxon, 2017-10-23:
http://prntscr.com/h147wl
Here is tag 
Nexus 7 tablet android 6

don, 2017-10-24:
The problem is your device doesn't support that tag. See the message "NXP Mifare Classic 1k - Not Supported" in your screenshot.

The 2nd generation Nexus 7 switched to a Broadcom NFC chip which doesn't support Mifare Classic NFC tags. The Nexus 7 should work find with any Type 1 through Type 4 NFC tags.

You can read the UID from the tag if you'd like. Add a Tag Discovered listener to index.js. Replace your `onDeviceReady` function with this one:

    onDeviceReady: function() {
        app.receivedEvent('deviceready');

        // Read NDEF formatted NFC Tags
        nfc.addNdefListener (
            function (nfcEvent) {
                var tag = nfcEvent.tag,
                    ndefMessage = tag.ndefMessage;

                // dump the raw json of the message
                // note: real code will need to decode
                // the payload from each record
                alert(JSON.stringify(ndefMessage));

                // assuming the first record in the message has
                // a payload that can be converted to a string.
                alert(nfc.bytesToString(ndefMessage[0].payload).substring(3));
            },
            function () { // success callback
                alert("Waiting for NDEF tag");
            },
            function (error) { // error callback
                alert("Error adding NDEF listener " + JSON.stringify(error));
            }
        );

        // Handle non-NDEF tags and Mifare Classic tags with Broadcom readers
        nfc.addTagDiscoveredListener (
            function (nfcEvent) {
                var tag = nfcEvent.tag;
                alert(JSON.stringify(ndefMessage));
            },
            function () { // success callback
                alert("Waiting for non-NDEF tags");
            },
            function (error) { // error callback
                alert("Error adding tag listener " + JSON.stringify(error));
            }
        );
    
    },

closed 2018-04-18 by don