chariotsolutions/phonegap-nfc #297: iOS 11 - timing out on reading

fairport, 2017-10-25:
Hi Don ...

I am a total NFC novice.
I get the impression that everything is working OK.. I see the screen telling me its ready to scan ...  
but I am not sure what type of RFID tags I should be able to read with iOS. I have tried a couple of different types that I have with me but the device (iPhone 7) times out.

We're developing this app for a client who will be putting tags on trees in an orchard & they are asking us for the tag types they'll need.
Could you point me in the right direction, please? Maybe an eBay listing perhaps. I have seen a number of different tag types there. 

Thanks
Roger 
Fairport

erikm30, 2017-10-25:
Please get in touch with Nfcapp.support@nxp.com.
We can guide you in the right direction.

don, 2017-10-27:
@fairport It's possible you have the correct tags, but there's no NDEF message on them. If you have an Android device available, try writing a message with [NFC TagWriter by NXP](https://play.google.com/store/apps/details?id=com.nxp.nfc.tagwriter&hl=en). You should be able to use any NFC type 1 through type 4 tags. I'd avoid Mifare Classic tags since they're not try NDEF tags.

For more details on the tags, I like @erikm30 suggestion to reach out to NXP nfcapp.support@nxp.com 

If you need more general NFC info, you could checkout [Beginning NFC](https://www.amazon.com/Beginning-NFC-Communication-Arduino-PhoneGap/dp/1449372066) some of PhoneGap info is dated, but a lot of the code and info is still good.

closed 2018-04-18 by don