chariotsolutions/phonegap-nfc #300: iPhone 7 can't get nfcEvent to read the code

pediroda, 2017-11-15:
plugin version : 0.7.1
BeginSession --> successful reading :  the console shows the value of the read tag (ndefMessage...) that's great.... BUT it's not from my own code (I assume that it is read in the plugin) !

I have a nfc.addNdefListener(onNfc, nfcSuccess, nfcFailure) so I expect my onNFC() function to catch the event and read the data as on Android. 

What's wrong ? How could I access the nfcEvent.tag ?

pediroda, 2017-11-16:
Hi, is my problem interesting for someone ;-) ?

HellHound90, 2017-11-17:
Hi @pediroda could you provide your code?

pediroda, 2017-11-18:
Hello, 
thank you
I have partially solved the issue . 
1) FIrst issue is that I never entered in onNfc()... it seems even that is did not enter into nfcEnabled(). I don't explain it but after different tests & "ionic build ios", I finally manage to get into onNfc().
2) tag.id seems 'undefined" in IoS. Thus nfc.bytesToHexString(tag.id) crashed the app
For the first point, i can't explain... for the second, it is surely a bug ?

Here is my code: 

Controller: 
ionic.Platform.ready(function () {
  if (window.cordova) {
     nfc.enabled(nfcEnabled, nfcDisabled);
  }
}

function nfcEnabled(result) {
 nfc.addNdefListener(onNfc, nfcSuccess, nfcFailure)
}

function onNfc(nfcEvent) {
   var tag = nfcEvcent.tag, 
            ndefMessagfe = tag.ndefMessage;

   console.log("Tag SN: " + tag.serialNumber;
   /*** CRASHES APP: console.log("TAD ID:  " + nfc.bytesToHexString(tag.id); ****/
   ndefMessage.forEach (function (record) {
       console.log (JSON.stringify(record.payload)
   }
}

HellHound90, 2017-11-18:
ndefMessagfe = tag.ndefMessage;

Maybe try to remove the “f” letter and let us know.

pediroda, 2017-11-19:
No, it's not the issue, it's just because I rewrited the code (no copy/paste) :-)

don, 2018-04-18:
iOS doesn't provide the tag id. You can see #287 for how to do get the id, but it's not being merged because apps will be rejected.

See https://github.com/don/phonegap-nfc-ios for an example of reading tags on iOS

closed 2018-04-18 by don