chariotsolutions/phonegap-nfc #301: Not returning any text p2p (phone to phone)

RicardoRibeirorr, 2017-12-05:
I'm using the p2p sample from your repository in one phone, in he other i'm using the read sample.
I have tryde all the possibilitis i have, all the samples and modification and even read you book like 3 times, and steel don't undertand.

Can anyone tell me if to do a peer to peer app, just sending a simple text from one phone to another, it's just need to use(nfc.share) and (nfc.read)?

AdrianMrn, 2018-01-22:
Did you manage to find a solution to this problem?

don, 2018-04-18:
Use [nfc.addNdefListener](https://github.com/chariotsolutions/phonegap-nfc#nfcaddndeflistener) to receive messages. A NDEF message sent from a peer works like reading a NFC tag.

You should be able to share a message from https://github.com/don/phonegap-p2p and receive it with https://github.com/don/phonegap-nfc-reader. (Those project are old so they might need to be updated.)

If you use nfc.share to send a NDEF message with a URI, many Android phones will receive the message without an app.

    var message = [
        ndef.uriRecord("https://cordova.io");
    ];

    nfc.share(message, [onSuccess], [onFailure]);

closed 2018-04-18 by don

erikm30, 2018-04-18:
To share data from Android to iOS with NFC, nfc.share doesn't work.
You need to implement a HCE application that responds to the few commands that are defined by Tag 4 Type (mainly read binary)

HerrZwerg, 2018-09-23:
Hi erikm30,
can you share some more info in this? maybe even a quick sample?

much appreciated