chariotsolutions/phonegap-nfc #305: How to receive data from NFC devices

AdrianMrn, 2018-01-22:
I'm trying to use one device to share a message to another device over p2p.

In the description of the repo, it says I can use this plugin to "receive data from NFC devices".

What function do I call to do this? I went through all the available functions and could only find the function nfc.share, which sends a message over p2p, but doesn't seem to receive one.

allannaranjo, 2018-01-22:
Seems like you have to use (as in the documentation)
var message = [
    ndef.textRecord("hello, world")
];

nfc.share(message, [onSuccess], [onFailure]);

That will appear as if you are reading a tag in the other device.

So I would say, you do nfc.share(...) then the other device once at the right distance will read that message, making your first device act like an NFC tag.

Only in:
Android
Windows
BlackBerry 7
BlackBerry 10
Windows Phone 8

don, 2018-04-18:
Use [nfc.addNdefListener](https://github.com/chariotsolutions/phonegap-nfc#nfcaddndeflistener) to receive messages. A NDEF message sent from a peer works like reading a NFC tag.

closed 2018-04-18 by don