chariotsolutions/phonegap-nfc #309: Methods of injected nfc object doesn't work

JonasReichhardt, 2018-02-27:
Hi i've got the problem that as soon as I'm using a method of the injected nfc object in my javascript the statement isn't executed along the statements behind.

I'm using Phonegap Build 0.4.5 and a Samsung Galaxy 5 mini with Android 6.0.1

Anyone got the same problem or can help me?

don, 2018-04-18:
"nfc.enabled is not a function" usually means the plugin is not installed correctly, but sometimes it's caused by a JavaScript error or build error.

Make sure the plugin is defined in config.xml. https://github.com/chariotsolutions/phonegap-nfc#phonegap-build

    <gap:plugin name="phonegap-nfc" source="npm" />

I don't have a good way to debug PhoneGap build. I'd recommend installing the Android SDK and doing local development with Cordova. This way you can use`adb logcat`to see Java errors and chrome://inspect to debug the Javascript on your phone from your computer.

JonasReichhardt, 2018-04-19:
I've already fixed the problem.

Unfortunately by changing to native Android development. We had a really short time window to finish the project therefore I needed to change technology.

But still thanks for your answer and please keep up the work on this plugin

Jonas Reichhardt

closed 2018-04-19 by JonasReichhardt