chariotsolutions/phonegap-nfc #310: NFC plugin stops working if NFC is turned on without application restart

Maduiini, 2018-03-14:
OS: Android 7.0

NFC plugin no longer reads tags if the phone's NFC is turned off and then on again (or was off and is turned on). This problem persists until app restart or if onResume event is triggered, for example by going to lockscreen or another app and back. (Visiting the phone's dropdown quickmenu doesn't bring the app into background.) Also if the nfc plugin wasn't initialized at all (nfc off when starting app) the only fix is to restart the application to init it.

nfc.enabled returns "NFC_OK" even though it doesn't read anything and the plugin's startNfc is not called before onResume.

Could there be a way to dynamically initialize the plugin if it wasn't on? Currently only init and onResume methods can call startNfc and it cannot be called any other way from the js side since it's protected.

don, 2018-04-18:
@Maduiini Is this still a problem? Are you seeing this on more than one device? I haven't seen this in the past, I can't replicate it on Samsung S8 with Android 8.0.0. (I don't have a Android 7.0 phone handy.)

Can you run with the AndroidStudio debugger and make sure [onResume()](https://github.com/chariotsolutions/phonegap-nfc/blob/7e1c24fc0f617c8b0866bfab63924908fbdefad0/src/android/src/com/chariotsolutions/nfc/plugin/NfcPlugin.java#L718) is getting called?

Maduiini, 2018-04-19:
OnResume() was only getting called when returning to the app from lockscreen/other apps. Not when visiting the dropdown-quickmenu.

Figured out a workaround by adding a BroadcastReceiver inside the plugin that listens to changes in the phone NFC and calls startNfc() / stopNfc(). This way it always changes the plugin state if NFC is turned off/on.

closed 2018-04-19 by Maduiini

KreoDev, 2019-01-02:
Hi there
I'm having the same problem on Android 7.0. Could I please have more information on how to create this workaround, I have never really modified a plugin before.