chariotsolutions/phonegap-nfc #312: Not work on iOS - iPhone 7 and more

Alex-Zayats-Noveo, 2018-04-11:
Did everything according to the instructions
On in Xcode - http://joxi.ru/zANG6gYUlbzxEr
Does not work reading and writing
NFC Peer to Peer

Here is an example code

Write - NFC offers a bar, but there is no connect

```
nfc.beginSession(function () {
                       console.log('start session');
                      nfc.share(message, onSuccessNFC, onFailureNFC);
                     }, function (err) {
                       console.log(err);
                       console.log('failed start session');
                     });
```

Read - NFC offers a bar, but there is no connect

```
nfc.beginSession(function () {
                       console.log('start session');
                       nfc.addNdefListener(
                         function (nfcEvent) {
                           var tag = nfcEvent.tag, я
                           ndefMessage = tag.ndefMessage;
                           var code = nfc.bytesToString(ndefMessage[0].payload).substring(3);
                           nfcCodeSuccess(code);
                         },
                         function () { // success callback
                           // alert("On NFC");
                         },
                         function (error) { // error callback
                           alert("Error adding NDEF listener " + JSON.stringify(error));
                         }
                       );
                     }, function () {
                       alert('failed start session')
                     });
```

don, 2018-04-17:
iOS doesn't support peer-to-peer. This needs to run on iPhone7 or higher.  The NFC tag needs to contain a NDEF message.

See if you can get this sample project to work https://github.com/don/phonegap-nfc-ios

devotoare, 2018-05-06:
@don are there plans to get this working with iOS peer-to-peer? Or is it not possible for a technical reason?

don, 2018-05-06:
@devoarte NFC on iOS is very limited due to Apple’s restrictions. It’s not possible to send peer to peer on iOS. iOS *might* be able to receive a peer to peer message from Android.

closed 2018-05-29 by don