chariotsolutions/phonegap-nfc #314: Error: invalid variable format: scanning when installing

jberke, 2018-04-17:
I am using VS2017 and try to install the plugin from GIT and I get this output

Installing plugin 'https://github.com/chariotsolutions/phonegap-nfc'

You have been opted out of telemetry. To change this, run: cordova telemetry on.
Error: invalid variable format: scanning
Done

don, 2018-04-17:
This sounds like a VS2017 problem. I don't use that tool, so I'm not sure.

I recommend installing the plugin into your project from npm using the Cordova tools

    cordova plugin add phonegap-nfc

jberke, 2018-04-18:
Yea it probably is, I did install the plugin with the CLI and its working like a champ! Nice plugin by the way

don, 2018-04-18:
Thanks for the update. Glad it's working.

closed 2018-04-18 by don