chariotsolutions/phonegap-nfc #317: nfc.addMimeTypeListener re-opens the app again (phonegap)

Farynx, 2018-05-19:
So I'm trying to send information from one phone to another through NFC using phonegap. I want whatever information I send from one app to be saved as a variable at some point so I can either display it, manipulate it, etc. The issue I'm having is that whenever I try to send a text from one phone, it just opens up the app but does not display the text like it should. I have a custom Mimetype and an intent filter set up to open up the app on the other phone, so I don't believe that to be the problem.I think I may be using the addMimeTypeListener function wrong but am not too sure. Any help in clarifying my issue would be appreciated.

Here is where I call the addMimeTypeListener function in the deviceready function:

    deviceready: function () {
    document.getElementById('checkbox').addEventListener('change', app.toggleCheckbox, false);
    sample.addEventListener('click', app.showSampleData, false);

    nfc.addNdefListener(
        function() {
            app.notifyUser("Found an NDEF formatted tag");
        },
        function() {
            app.notifyUser("Success.");
        },
        function() {
           app.notifyUser("Fail.");
        }
    );
    nfc.addMimeTypeListener(
        "text/shunt",
        parseTag,

        function() {
           app.notifyUser("Success");
        },
        function() {
            app.notifyUser("Fail");
        }
    );
  },

Here is where call the parsed tag function

    function parseTag(nfcEvent) {
var records = nfcEvent.tagData;

for (var i = 0; i < records.length; i++) {
    var record = records[i],
        p = document.createElement('p');
    p.innerHTML = nfc.bytesToString(record.payload);
    display.appendChild(p);
    app.notifyUser(p);
}

And here is my intent filter just for reference

    <application android:hardwareAccelerated="true" android:icon="@mipmap/icon" android:label="@string/app_name" android:supportsRtl="true">
    <activity android:configChanges="orientation|keyboardHidden|keyboard|screenSize|locale" android:label="@string/activity_name" android:launchMode="singleTop" android:name="MainActivity" android:theme="@android:style/Theme.DeviceDefault.NoActionBar" android:windowSoftInputMode="adjustResize">
        <intent-filter android:label="@string/launcher_name">
            <action android:name="android.intent.action.MAIN" />
            <category android:name="android.intent.category.LAUNCHER" />
        </intent-filter>
        <intent-filter>
            <action android:name="android.nfc.action.NDEF_DISCOVERED" />
            <data android:mimeType="text/shunt" />
            <category android:name="android.intent.category.DEFAULT" />
        </intent-filter>
    </activity>
</application>

I've been trying to follow chariotsolutions examples in order to understand how it works but still not having much luck.

Thank you.

don, 2018-05-29:
@Farynx in the past adding `android:noHistory="true"` to the activity xml would stop Android from launching a new app. I'm not sure it it still works.

don, 2018-06-08:
@Farynx did adding those flags help?

Farynx, 2018-06-11:
@don we ended up combining your reading nfc code and your peer to peer nfc code and it worked. We did not try changing the xml file. Thanks anyway for the suggestion.

don, 2018-06-11:
@Farynx sounds good. Let me know if there's any documentation or code that needs to be updated.

closed 2018-06-11 by don