chariotsolutions/phonegap-nfc #318: Is it possible make this work to read EMV cards? using this library devnied/EMV-NFC-Paycard-Enrollment

itsvicsoto, 2018-05-24:
Hi guys, I am trying to this api to read nfc and I did get some tags but I'm not sure if I can use those tags to use it on this library devnied/EMV-NFC-Paycard-Enrollment. 

Another workaround is to use cordova plugin and start an activity to read nfc but I really want to avoid that route since i don't really want to start an activity just to read nfc. It would be better if I integrate the library using the native ionic NFC api.

don, 2018-05-29:
Right now this plugin only handles NDEF messages. You could look at https://github.com/don/cordova-plugin-hce for doing host card emulation. If you have a Java library that does payment processing, the best solution is to create new plugin that wraps your library.

mohammedgaus, 2020-06-24:
Can someone please help me in following

mohammedgaus, 2020-06-24:
@don  Can youplease help me in this
I want to implement payment option using nfc in my cordova application(need steps)
Also I need help for virtual testing this while development

MooseTheCoder, 2020-07-14:
@mohammedgaus I've just spent the last 10~ hours reading up on the EMV spec, it is possible to read card numbers and expiry info using this lib if you have the correct APDU commands.

I have not found an easy way to emulate an EMV card in a sandbox environment, so I've just been making builds and using my phone.

Start with this.

```js
const GET_PSE = '00 A4 04 00 0e 32 50 41 59 2e 53 59 53 2e 44 44 46 30 31 00';
```

To get the PSE, this will contain the AID.

Work though this then use `00 A4 04 00` with the AID you got to select the Application.

(I skipped the GPO step, but apparently you should put it in here).

Use this command `00 B2 01 14 00` to read the record. The card number is stored on tag `5a`.

This is a useful resource : https://www.eftlab.com/knowledge-base/145-emv-nfc-tags/

I hope this helps get you moving in the right direction!