chariotsolutions/phonegap-nfc #325: Uncaught Error: Module phonegap-nfc.NFC does not exists.

ericaig, 2018-06-06:
I'm not able to use the app after updating to the newer version (1.0.1)

In console, I see this error:
Uncaught SyntaxError: Unexpected strict mode reerved word
Uncaught Error: Module phonegap-nfc.NFC does not exists.

don, 2018-06-06:
> Uncaught SyntaxError: Unexpected strict mode reserved word
Some of the new Javascript code uses let, const, and arrow functions

What version of Cordova are you using? What version of Android are you deploying to?
When do you see this message, during Cordova build?

ericaig, 2018-06-07:
Hey @don thanks for your response. I updated the plugin to the latest version 1.0.2 but it still doesn't work. Although the error kindda changed a wee bit. With version 1.0.1, I get:

Uncaught SyntaxError: Unexpected strict mode reerved word 
Uncaught Error: Module phonegap-nfc.NFC does not exists.

And with version 1.0.2, I get:
Uncaught SyntaxError: Unexpected identifier
plugins/phonegap-nfc/www/phonegap-nfc.js:517
Uncaught Error: Module phonegap-nfc.NFC does not exist.

[image]

I'm using:

cli packages: (/usr/local/lib/node_modules)

     @ionic/cli-utils  : 1.19.2
     ionic (Ionic CLI) : 3.20.0

global packages:

     cordova (Cordova CLI) : 8.0.0

local packages:

     @ionic/app-scripts : 3.1.9
     Cordova Platforms  : android 7.0.0
     Ionic Framework    : ionic-angular 3.9.2

don, 2018-06-08:
@vpedrosa are you using Ionic v1 also?

@ericaig I haven't had a chance to debug this yet, but I'm guessing for the Ionic v1, there's some additional step or flag needed to cross compile the new ES6. It's also possible there are some new Android API calls that break old stuff.

If either of you have time to track this down, please add more info here.

don, 2018-06-11:
@ericaig @vpedrosa  I updated my Ionic v1 example so it runs with the Ionic 3.20.0 command line tools. It didn't break for me. Can you see if you can run this project? https://github.com/don/ionic-nfc-reader

don, 2018-06-12:
I found an old phone with Android 4.4 and can duplicate the problem. It's because of the new ES6 syntax in phonegap-nfc.js. You can either add webpack to your build process or wait for the next release. I've converted everything back to ES5, but I need to do more testing before I can release.

ericaig, 2018-06-12:
I'm using Ionic 3.20.0. I've been running some tests myself and I had to downgrade the plugin back to 0.7.3.

Everything works fine... but :'( I can't "deactivate" NFC after reading or writing to an NFC tag. It continues to execute NFC event even after leaving the NFC  read/write page.

Is there a way I can deactivate and reactivate it whenever it's needed?

Thanks @don

don, 2018-06-12:
@ericaig I recommend ignoring the nfcEvent in your callback if you don't need to process it. There are also functions to remove listeners e.g. nfc.removeNdefListener

closed 2018-06-12 by don

don, 2018-06-12:
@ericaig if you're just reading tags, look at the [nfc.readerMode()](https://github.com/chariotsolutions/phonegap-nfc#nfcreadermode). There's no Ionic wrappers, I'm working on them today.

ericaig, 2018-06-14:
Yeeees!! everything is working great... instead of using nfc.addNdefListener, I used nfc.readerMode() as you suggested and nfc.disableReaderMode do deactivate it.

Cheers :clinking_glasses:  @don

ericaig, 2018-06-14:
For anybody that needs help with "writing", I suggest you check this out..

https://github.com/chariotsolutions/phonegap-nfc/issues/328

commit e6517e10 (2018-06-12): Remove ES6 syntax from phonegap-nfc.js
Fixes #325