chariotsolutions/phonegap-nfc #327: TypeError Cannot read property 'length' of undefined

annYang66, 2018-06-13:
having this error while build app to my android phone(using Visual Studio 2015)

> Warning: Failed to install 'phonegap-nfc':TypeError: Cannot read property 'length' of undefined
>     at myCordovaApp\platforms\android\cordova\node_modules\cordova-common\src\ConfigChanges\ConfigChanges.js:381:23
>     at Array.forEach (native)
>     at is_conflicting (myCordovaApp\platforms\android\cordova\node_modules\cordova-common\src\ConfigChanges\ConfigChanges.js:355:17)
>     at PlatformMunger.add_plugin_changes (myCordovaApp\platforms\android\cordova\node_modules\cordova-common\src\ConfigChanges\ConfigChanges.js:150:33)
>     at myCordovaApp\platforms\android\cordova\node_modules\cordova-common\src\PluginManager.js:126:25
>     at _fulfilled (myCordovaApp\platforms\android\cordova\node_modules\q\q.js:834:54)
>     at self.promiseDispatch.done (myCordovaApp\platforms\android\cordova\node_modules\q\q.js:863:30)
>     at Promise.promise.promiseDispatch (myCordovaApp\platforms\android\cordova\node_modules\q\q.js:796:13)
>     at myCordovaApp\platforms\android\cordova\node_modules\q\q.js:857:14
>     at runSingle (myCordovaApp\platforms\android\cordova\node_modules\q\q.js:137:13)
> C:\Users\123\AppData\Roaming\npm\node_modules\vs-tac\node_modules\q\q.js:126
>                     throw e;
>                           ^
> TypeError: Cannot read property 'length' of undefined
>     at myCordovaApp\platforms\android\cordova\node_modules\cordova-common\src\ConfigChanges\ConfigChanges.js:381:23
>     at Array.forEach (native)
>     at is_conflicting (myCordovaApp\platforms\android\cordova\node_modules\cordova-common\src\ConfigChanges\ConfigChanges.js:355:17)
>     at PlatformMunger.add_plugin_changes (myCordovaApp\platforms\android\cordova\node_modules\cordova-common\src\ConfigChanges\ConfigChanges.js:150:33)
>     at myCordovaApp\platforms\android\cordova\node_modules\cordova-common\src\PluginManager.js:126:25
>     at _fulfilled (myCordovaApp\platforms\android\cordova\node_modules\q\q.js:834:54)
>     at self.promiseDispatch.done (myCordovaApp\platforms\android\cordova\node_modules\q\q.js:863:30)
>     at Promise.promise.promiseDispatch (myCordovaApp\platforms\android\cordova\node_modules\q\q.js:796:13)
>     at myCordovaApp\platforms\android\cordova\node_modules\q\q.js:857:14
>     at runSingle (myCordovaApp\platforms\android\cordova\node_modules\q\q.js:137:13)

im using 
`Cordova cli 6.3.0`

while use config.xml to install phonegap nfc plugin in VS 2015, it shows a column `NFC_USEAGE_DESCRIPTION`,
is it needed to fill?
can't find what to fill in README.

don, 2018-06-13:
I need to document NFC_USAGE_DESCRIPTION. It's for iOS to display to the user when requesting NFC permissions.

If you specify NFC_USAGE_DESCRIPTION when installing the plugin

    cordova plugin add phonegap-nfc --variable NFC_USAGE_DESCRIPTION="Scan NFC Tags"

It adds this info into config.xml

    <plugin name="phonegap-nfc" spec="^1.0.3">
        <variable name="NFC_USAGE_DESCRIPTION" value="Scan NFC Tags" />
    </plugin>

I'm not sure what's causing the stack trace. I tried downgrading to cordova@6.3.0 and test on my mac, but the old version won't run without downgrading Gradle.

Suggestions
 * Upgrade to cordova cli 8.0.0 `npm install -g cordova` 
 * Make sure you're running Nodejs 8.11.x
 * Upgrade the native code to cordova-android-7.0.0 (use cordova to remove and install the platform)

If you can't upgrade Cordova, see if an older version of the plugin works better

    cordova plugin rm phonegap-nfc
    cordova plugin add phonegap-nfc@0.7.3

Let me know if any of that works (or doesn't).

annYang66, 2018-06-14:
i use cordova plugin add phonegap-nfc@0.7.3 and it works. thx!

closed 2018-06-14 by don