chariotsolutions/phonegap-nfc #328: App closes when writing to an NFC tag

ferri-17, 2018-06-13:
Hello Don, I updated the plugin to the latest version 1.0.3, but i have a problem, my app reads and write to an NFC tag. Reading and writing works correctly, but when I exit the read/write page and goes back on it to write to an NFC tag,  the app closes automatically.

**_My code:_**

init()  {

    setTimeout(function () {
        
      this.nfc.addNdefListener(
        (rnfcevent) => {
            console.info("*****Second param callback nfc", rnfcevent);

          var message = this.ndef.textRecord('hello world');
            this.nfc.write([message]);

          let tagId = this.nfc.bytesToHexString(rnfcevent.tag.id);
          console.log('UID',tagId);
            
          this.nfc.removeNdefListener(() => { }, () => {
              console.info("remove success nfc");
            });
          }, () => {
            console.info("*****Success. nfc");
          }, () => {
          console.error("*****Error. nfc");
        }, (stat) => {
            console.log('staus',stat);
        })
        }, () => {
        console.error("*****remove mimetypelistener nfc error");
    }, 10);

**_I am using:_**

cli packages: (/usr/local/lib/node_modules)

    @ionic/cli-utils  : 1.19.2
    ionic (Ionic CLI) : 3.20.0

global packages:

    cordova (Cordova CLI) : 8.0.0

local packages:

    @ionic/app-scripts : 3.1.8
    Cordova Platforms  : android 7.0.0 ios 4.5.4
    Ionic Framework    : ionic-angular 3.9.2

don, 2018-06-13:
Here's the stack trace
```
06-13 11:49:29.594  7832  7938 E AndroidRuntime: FATAL EXCEPTION: pool-1-thread-2
06-13 11:49:29.594  7832  7938 E AndroidRuntime: Process: com.megster.nfc.issue328, PID: 7832
06-13 11:49:29.594  7832  7938 E AndroidRuntime: java.lang.IllegalStateException: Close other technology first!
06-13 11:49:29.594  7832  7938 E AndroidRuntime: 	at android.nfc.Tag.setConnectedTechnology(Tag.java:458)
06-13 11:49:29.594  7832  7938 E AndroidRuntime: 	at android.nfc.tech.BasicTagTechnology.connect(BasicTagTechnology.java:78)
06-13 11:49:29.594  7832  7938 E AndroidRuntime: 	at android.nfc.tech.Ndef.connect(Unknown Source:0)
06-13 11:49:29.594  7832  7938 E AndroidRuntime: 	at com.chariotsolutions.nfc.plugin.NfcPlugin.lambda$writeNdefMessage$2$NfcPlugin(NfcPlugin.java:344)
06-13 11:49:29.594  7832  7938 E AndroidRuntime: 	at com.chariotsolutions.nfc.plugin.NfcPlugin$$Lambda$2.run(Unknown Source:6)
06-13 11:49:29.594  7832  7938 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
06-13 11:49:29.594  7832  7938 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
06-13 11:49:29.594  7832  7938 E AndroidRuntime: 	at java.lang.Thread.run(Thread.java:764)
```

I think this is because you're removing the NDEF listener in the event handler. Adding and removing listeners changes recently and restarts nfc as part of the process, so code that worked before might fail now.

Can you see if this works better?

```
import { Component } from '@angular/core';
import { IonicPage, NavController, NavParams } from 'ionic-angular';
import { NFC, Ndef } from '@ionic-native/nfc';
import { Subscription } from 'rxjs/subscription';

@IonicPage()
@Component({
  selector: 'page-nfc',
  templateUrl: 'nfc.html',
})
export class NfcPage {

  subscription:Subscription;

  constructor(public navCtrl: NavController,  private nfc: NFC, private ndef: Ndef) {

  }

  ionViewWillEnter() {

    // long form, handles the promise when adding the listener
    this.subscription = this.nfc.addNdefListener(
      () => console.log('Added NDEF Listener'),
      error => console.log('Error adding NDEF listener')
    ).subscribe(this.onNfcEvent.bind(this));

    // optionally you can skip the promise and just subscribe
    //this.subscription = this.nfc.addNdefListener().subscribe(this.onNfcEvent.bind(this));

  }

  onNfcEvent(nfcEvent) {
    console.log('NFC Event', nfcEvent);

    let tagId = this.nfc.bytesToHexString(nfcEvent.tag.id);
    console.log('UID',tagId);

    var message = this.ndef.textRecord('hello, world', 'en', null);
    this.nfc.write([message]).then(
      () => console.log('Wrote data to NFC tag'),
      error => console.log('Error writing to NFC tag', error)
    )
  }

  ionViewWillLeave() {
    console.log('Leaving view. Removing NDEF subscription');
    // unsubscribe so tags don't scan on other pages
    // unsubscribing calls ble.removeNdefListener    
    this.subscription.unsubscribe();
  }

}
```

ferri-17, 2018-06-14:
Hey @doncoleman, thank you for answering, I don't have` adb logcat`, but I tried your code, and work fine, thank you for everything. I'll close this problem.

closed 2018-06-14 by ferri-17

hamdimefteh, 2021-06-24:
let flags = this.nfc.FLAG_READER_NFC_A | this.nfc.FLAG_READER_NFC_V;
    this.nfc.readerMode(flags).subscribe(

      tag => {
        alert(JSON.stringify(tag))
        // this.nfc.connect('android.nfc.tech.MifareClassic', 3000).then(
        //   () => {

        // this.nfc.addNdefListener(() => {
        //   //this.writetag();

        //   this.presentAlert('ok');
        //   //this.displayNdef;
        // }, (err) => {
        //   this.presentAlert('ko' + err);
        // }).subscribe( () => {
        //   alert('debut');

        //   this.writetag();

        //   alert("fin");
        //   // alert(event.tag.id);
        // }
        //   , err => alert(err));
        this.subscription = this.nfc.addNdefListener(
          () => alert('Added NDEF Listener'),
          error => alert('Error adding NDEF listener')
        ).subscribe(this.onNfcEvent.bind(this));
      });

    //   },
    //   (error) => alert('connection failed' + error)

    // )

    err => alert('Error reading tag' + err)

    this.nfc.close;

  }

hamdimefteh, 2021-06-24:
read work but write d'ont work watever i did . who could help please ?!