chariotsolutions/phonegap-nfc #329: build failure

pignattino, 2018-06-14:
I'm trying to build a project with your plugin (using jdk 1.8) and I get these errors:

**_.......platforms/android/src/com/chariotsolutions/nfc/plugin/NfcPlugin.java:80: error: diamond operator is not supported in -source 1.6
    private final List<IntentFilter> intentFilters = new ArrayList<>();_**

and

_**.......platforms/android/src/com/chariotsolutions/nfc/plugin/NfcPlugin.java:214: error: lambda expressions are not supported in -source 1.6
        getActivity().runOnUiThread(() -> {**_

Any help?
Thanks a lot

don, 2018-06-14:
Can you update your Android project to use source 1.8?

pignattino, 2018-06-14:
using old version (0.7.1) works.. I don't understand how to update Android project, it's created by cordova

don, 2018-06-14:
0.7.3 will work too 

If you haven’t made any modifications to the generated Android project, removing and installing the android platform from your app should let you use the latest version of the plugin. Assuming you use Cordova 8.0.0 with cordova-android-7.0.0.

pignattino, 2018-06-15:
ok removing and adding again the Android platform I can use version 1.0.3
Thanks a lot Don!

pignattino, 2018-06-15:
another question Don.. 
Plugin works, but if I see console it says

_**Uncaught ReferenceError: PhoneGap is not defined
    at phonegap-nfc.js:3**_

any javascript loading sequence problem? cordova.js is the first loaded..
Thanks again

don, 2018-06-19:
I haven't seen that message. Maybe something happened with the module loading? Maybe it's related to the old init code and the new channel code? Make sure you wait for device ready before calling any nfc functions.  I'm going to be on vacation for a week so it'll be a while before I can look at it. Since the plugin works, hopefully you can work around it for now.

closed 2019-10-18 by don