chariotsolutions/phonegap-nfc #330: When subscribed to both nfc.addTagDiscoveredListener and nfc.addMimeTypeListener - discover doesn't fire

nKittie, 2018-06-17:
Hi guys, thank you for the awesome library. I'm trying to make use of it to implement a reader/writer feature on out JS web app and I'm having following issues - could you be so kind as to advise?

I have followed your individual sample projects code.

When I have initialised both event listeners nfc.addTagDiscoveredListener and nfc.addMimeTypeListener - I only see the MimeTypeListener firing so I'm handling the reads of the tag just fine but I do not detect TagDiscoveredListener listener firing whatsoever. When only nfc.addTagDiscoveredListener is initialised it fires as expected.

My 1-page app has a read/write mode switcher, so when both listeners are initialised I'd expect both of them to fire and my app to determine which listener to follow up. As a workaround I wanted to use the TagDiscoveredListener to display the tag data, but looking at the nfcEvent it does not even have NdefMessage - should that be present?

Could you kindly let me know what the issue with TagDiscoveredListener not firing might be? Would you be able to provide a sample project which does both reading and writing on a single page?

Thank you very much.

don, 2018-06-19:
When you have [multiple listener](https://github.com/chariotsolutions/phonegap-nfc#multiple-listeners) Android only ever calls the most specific one. If there's tag discovered and mime type listeners, the mime type listener handles mime type tags. If you scan an NDEF tags with TNF=WellKnown, the tag discovered listener will handle that. If you add a NDEF listener, the NDEF listener would get the TNF=WellKnown instead of the tag discovered listener. It's a bit odd but that's the way Android works.

TagDiscoveredListener doesn't return NDEF data, try addNdefListener instead.

It depends on what your app is doing. If your app handles NDEF tags with mime media messages, use a mime type listener. If you want to overwrite any other tags with mime media messages, you can add a TagDiscoveredListener to do this. It should get NDEF, non-NDEF, and NDEF-Formatable tags. Alternately you could have one function handle a bunch of the listeners and let your event listener look at the [event type](https://github.com/chariotsolutions/phonegap-nfc#types) so see how to handle it.

You might want to take a look at the new [nfc.readerMode](https://github.com/chariotsolutions/phonegap-nfc#reader-mode-functions) to see if that works for your use case.

I don't have an example project with reading and writing. NFC always reads. I usually check a flag in my app to see if I should be writing.

nKittie, 2018-06-19:
Thank you very much - got it working using a combination of NFC/Mime/TagDetected listeners now.

closed 2018-06-19 by nKittie