chariotsolutions/phonegap-nfc #332: Error Build Cordova - error: diamond operator is not supported in -source 1.6

luizalexandrew, 2018-07-04:
I'm trying to build using version 1.0.3 of the plugin and is giving this error
[image]

cordova 7.1.0
android-platform 6.4.0
java version "1.8.0_171"

I downgraded to version 0.7.3 and this version is working

VolodymyrAndrushchak, 2018-07-05:
@luizalexandrew 
Just change JAVA_VERSION on 1_8 in your gradle file

don, 2018-07-06:
@luizalexandrew, did @VolodymyrAndrushchak 's suggestion work for you?

Thayakaran, 2018-07-12:
@VolodymyrAndrushchak where to find gradle file in Ionic 2. 
i'm new to ionic.
Thanks in advance.

tylervz, 2018-07-16:
@VolodymyrAndrushchak while that may be a fix for local builds, all cloud builds from a service such as Adobe PhoneGap Build will fail.

peikezot, 2018-07-30:
Did anyone find a workaround for the cloud builds? I'm using the cloud service from Adobe's build.phonegap.com and it is indeed complaining about the diamond operators and the lambdas.

fsobarzo, 2018-08-24:
@VolodymyrAndrushchak JAVA_VERSION on 1_8 need Jack, but this is deprecated

https://source.android.com/setup/build/jack

> Warning: As of this March 14, 2017 announcement, the Jack toolchain is deprecated. Jack was the default Android build toolchain for Android 6.0–8.1.

sabridundar45, 2018-08-24:
https://www.youtube.com/watch?v=AQ3-i4BqOrA

fsobarzo, 2018-08-28:
I added both parameters into `build.gradle` then do the build, but I got this error:

```
Could not find method enable() for arguments [true] on JackOptions_Decorated{isEnabled=null, isJackInProcess=null, additionalParameters=null} of type com.android.build.gradle.internal.dsl.JackOptions.
```

I have JAVA_VERSION 1_8.

fsobarzo, 2018-08-28:
@luizalexandrew you can find a solution ?

fredmaggiowski, 2018-09-18:
Since I couldn't upgrade Java compiler version in my cordova-based project I've had to fork the project and patch the code in order to support older java 1.6.

[Here the diff](https://github.com/chariotsolutions/phonegap-nfc/compare/master...fredmaggiowski:master).

If you're interested in the changes I could clean up the code a bit and make a pull request.

don, 2018-10-09:
The best fix is to change the compiler to use a newer language level. Alternately you can try and older version of the plugin without the newer language features.

adibisna, 2018-11-07:
> @luizalexandrew
> Just change JAVA_VERSION on 1_8 in your gradle file

it works for me. thanks

KaiBirkenstock, 2018-12-04:
use this fork: https://github.com/fredmaggiowski/phonegap-nfc

closed 2018-12-11 by luizalexandrew

kumquat601, 2019-01-15:
@fredmaggiowski 's hack working.

at least, build successful.