chariotsolutions/phonegap-nfc #333: ionic 3 + phonegap-nfc 1.0.3 + ios 11.4 can not work

csky89, 2018-07-16:
### android is very good, but ios 11.4 is not work, is there any demo used ionic 3?

csky89, 2018-07-18:
My colleague took a long time to finally discover the cause of the problem.
### Now ionic-native(version 4.9.2) still does not support the nfc function of ios. You can not add "import ionic-native/nfc", only need add plugin phonegap-nfc.

Below is the code
1. in config.xml
[image]

2. 
[image]
[image]

closed 2018-07-18 by csky89

acilryandi, 2018-11-21:
what about the function of tag tech? ionic-native/nfc still not support for few function like connect, close, transceive