chariotsolutions/phonegap-nfc #334: Cannot use the plugin without ionic-native

KostasGeor, 2018-07-19:
I am trying to run the phonegap plugin without using the ionic-native. The problem is that I declare in my home.ts file variables (cordova:any, NFC:any) but none seems to work and I cannot call any function for my NFC App. What should I declare and how shall I use the plugin in order to call a function??

P.s I am using Ionic 3.

don, 2018-07-19:
The global variable for the plugin is lowercase nfc (not NFC.)

KostasGeor, 2018-07-23:
Thank you very much! :)

Dehfw, 2019-07-29:
@KostasGeor 
can u provide some demo code? (import lib and use a function form phonegap-nfc)
I'm struggling at using phonegap-nfc without ionic-native too

don, 2019-10-18:
https://github.com/chariotsolutions/phonegap-nfc#sample-projects

closed 2019-10-18 by don