chariotsolutions/phonegap-nfc #340: Android: Preventing default NFC app from taking away active focus from my app

kgillespieatmosphere, 2018-08-08:
I have an app using the NFC plugin and it works fine, however I have an issue on some Android devices the default Android NFC app will steal focus away from my app. Such as the Samsung Galaxy 8 and 7 as well as the Google Pixel will deactivate my app and pull open a generic NFC tag reader app. It's really fustrating.

Was wondering if there is a way to prevent this from happening. Essentially while my app is active I do not want any other apps coming up because an NFC tag is read. Seems like my only solution is to ask the user to disable their default NFC app but that doesn't see, right. I figure there must be some setting I'm missing in the config.xml or android manifest to do this.

Sorry if this is a dumb question but I just can't seem to find any information on this for Android and my config.xml looks like it matches with what the example projects are doing. Would appreciate even a hint on where I should look for more information.

kgillespieatmosphere, 2018-08-09:
I have read that you can get all NFC tags if you set the intent filters to null, null as described in this post. 

https://stackoverflow.com/questions/12620880/new-tag-collected-instead-of-reading-tags-of-application-nfc-android

How ever it looks like there is a specific if statement in the android code to prevent this. Why is this the case? How can I set the intent filter to wild cards?

https://github.com/chariotsolutions/phonegap-nfc/blob/master/src/android/src/com/chariotsolutions/nfc/plugin/NfcPlugin.java#L537

You can see here the plugin actually prevents the user from being able to do this.

Wade-BuildOtto, 2018-09-10:
having the same issue. scans and loads the tag manager

m4rcosx, 2018-09-12:
Same problems here... I notice that this problems occurs on devices that has Android Oreo (8.0). Does anyone notice that?

kgillespieatmosphere, 2018-09-14:
I have spotted it on a google pixel 1 running Android 9.0. We test on a lot of Android devices and it doesn't seem to be a sure fire issue for any one particular version, but more to do with the version of the tag manager app that is installed.

don, 2018-10-09:
If your app is open and scanning NFC tags, Android may open another NFC app if your app doesn't handle that tag type. You can try adding additional listeners like [nfc.addTagDiscoveredListener](https://github.com/chariotsolutions/phonegap-nfc#nfcaddtagdiscoveredlistener) to handle some of these cases. You can add as many listeners as you want, and Android will only call the most specific one.  Since you're seeing this on some devices and not others, it might be an Android anomaly, bug, or other annoying behavior.

You might have better luck with the newer Reader Mode API that is designed for foreground scanning. See [nfc.readerMode](https://github.com/chariotsolutions/phonegap-nfc#nfcreadermode) for more details.

kgillespieatmosphere, 2018-10-11:
Thanks @don, I'll give the addTagDiscoveredListener a shot and see if that helps with our issues. If not I'll move to the reader mode api and see if that changes anything.

fixdaddy, 2019-10-09:
I tried addTagDiscoveredListener, nfc.readerMode and yet I have the same problem. The default phone app is reporting the type T, known type NDEF tag and the developer's Phonebuild app is not regitering the event. I'm using [these](https://www.amazon.ca/Ntag215-NFC-Stickers-Tagmo-Amiibo/dp/B07N66X98B/ref=sr_1_1?keywords=ndef+tag&qid=1570663991&s=electronics&sr=1-1) tags, programmed with NFC Tools. I'm using a Huawei P20 phone. 

I have followed the "Getting Started CLI" steps to build the app. Just to be sure, the index.js file is as follows after the modification suggested in the guide:

var app = {
    // Application Constructor
    initialize: function() {
        this.bindEvents();
    },
    // Bind Event Listeners
    //
    // Bind any events that are required on startup. Common events are:
    // 'load', 'deviceready', 'offline', and 'online'.
    bindEvents: function() {
        document.addEventListener('deviceready', this.onDeviceReady, false);
    },
    // deviceready Event Handler
    //
    // The scope of 'this' is the event. In order to call the 'receivedEvent'
    // function, we must explicitly call 'app.receivedEvent(...);'
    onDeviceReady: function() {
    app.receivedEvent('deviceready');

    // Read NDEF formatted NFC Tags
    nfc.addNdefListener(
        function (nfcEvent) {
            var tag = nfcEvent.tag,
                ndefMessage = tag.ndefMessage;

            // dump the raw json of the message
            // note: real code will need to decode
            // the payload from each record
            alert("NDS1 "+ JSON.stringify(ndefMessage));

            // assuming the first record in the message has
            // a payload that can be converted to a string.
            alert("NDS2" + nfc.bytesToString(ndefMessage[0].payload).substring(3));
        },
        function () { // success callback
            alert("Waiting for NDEF tag");
        },
        function (error) { // error callback
            alert("Error adding NDEF listener " + JSON.stringify(error));
        }
    );
},
    // Update DOM on a Received Event
    receivedEvent: function(id) {
        var parentElement = document.getElementById(id);
        var listeningElement = parentElement.querySelector('.listening');
        var receivedElement = parentElement.querySelector('.received');

        listeningElement.setAttribute('style', 'display:none;');
        receivedElement.setAttribute('style', 'display:block;');

        console.log('Received Event: ' + id);
    }
};

Am I missing something?

fixdaddy, 2019-10-10:
Update to my previous post.

The PhoneGap Developer app was probably restricting the access.

I had to manually transfer the APK package to my phone and install it. All of the automatic methods I read worked for me. There were a lot of hurdles in making this possible.

After hours and hours of troubleshooting, I get to see the messages in the Getting Started App. Wow!

closed 2019-10-18 by don

tstone84, 2020-10-13:
I still see this issue on an Android Samsung S8 using readerMode. This issue was closed, yet no one posted a solution. Can someone please provide their solution to prevent the default Android NFC screen from taking focus? Thanks!

tstone84, 2020-10-14:
The solution to the above issue we were seeing was caused by converting the Observable to a promise. When reverting back to an observable, everything started working fine.