chariotsolutions/phonegap-nfc #341: nfc is not defined error

MaruksS, 2018-08-20:
I'm making an app with Ionic v1 using phonegap-nfc plugin.

I was following documentation at the installation, but I keep getting the same error over and over again.
I am testing app on a real device with NFC support and I can successfully launch and use [this project](https://github.com/don/ionic-nfc-reader) which uses the same plugin.
I have all the same plugins installed (including cordova-plugin-whitelist).

I get no other errors except `ReferenceError: nfc is not defined` .
I will include some files which may contain important information to help you understand the issue.

**config.xml**

<content original-src="index.html" src="http://192.168.1.100:8100" />

    <access origin="*" />
    <allow-navigation href="http://ionic.local/*" />
    <allow-intent href="http://*/*" />
    <allow-intent href="https://*/*" />
    <allow-intent href="tel:*" />
    <allow-intent href="sms:*" />
    <allow-intent href="mailto:*" />
    <allow-intent href="geo:*" />
    <preference name="webviewbounce" value="false" />
    <preference name="UIWebViewBounce" value="false" />
    <preference name="DisallowOverscroll" value="true" />
    <preference name="android-minSdkVersion" value="19" />
    <preference name="android-targetSdkVersion" value="26" />
    <preference name="BackupWebStorage" value="none" /> 

<plugin name="cordova-plugin-console" spec="^1.0.5" />

    <plugin name="cordova-plugin-network-information" spec="^1.3.3" />
    <plugin name="cordova-plugin-splashscreen" spec="^4.0.3" />
    <plugin name="ionic-plugin-keyboard" spec="^2.2.1" />
    <allow-navigation href="http://192.168.1.100:8100" />
    <plugin name="cordova-plugin-ionic-webview" spec="^2.0.3" />
    <plugin name="cordova-plugin-whitelist" spec="^1.3.3" />
    <plugin name="cordova-plugin-device" spec="^2.0.2" />
    <plugin name="phonegap-nfc" spec="^1.0.3">
        <variable name="NFC_USAGE_DESCRIPTION" value="Read NFC Tags" />
    </plugin>
    <engine name="ios" spec="^4.4.0" />
    <engine name="android" spec="^7.0.0" />

**index.html**

```
<meta name="viewport" content="viewport-fit=cover, initial-scale=1, maximum-scale=1, user-scalable=no, width=device-width">
<meta http-equiv="Content-Security-Policy" content="default-src *; style-src 'self' 'unsafe-inline'; script-src 'self' 'unsafe-inline' http://localhost:35729 https://api.*apilink*.com http://localhost:8000 https://code.ionicframework.com ">
 <script src="lib/ionic/js/ionic.bundle.js"></script>
    <script src="lib/ionic/js/jquery-3.2.1.min.js"></script>
    <script src="lib/ionic/js/moment.min.js"></script>
    <script src="lib/ionic/js/ng-cordova.min.js"></script>
    <script src="lib/ng-lodash/build/ng-lodash.min.js"></script>
    <script src="lib/angular-simple-calendar/dist/angular-simple-calendar.js"></script>
    <script src="lib/angular-translate/angular-translate.min.js"></script>
    <script src="lib/angular-translate-loader-static-files/angular-translate-loader-static-files.min.js"></script>
    <script src="lib/restangular/dist/restangular.min.js"></script>
    <script src="lib/socket.io-1.4.5.js"></script>

    <script src="cordova.js"></script>  
    <script src="js/app.js"></script>
```

**app.js**

```
angular.module('starter', ['ionic',
..... ])

.factory('nfcService', function ($rootScope, $ionicPlatform) {
        var tag = {};
        $ionicPlatform.ready(function() {
            nfc.addNdefListener(function (nfcEvent) {
                console.log(JSON.stringify(nfcEvent.tag, null, 4));
                $rootScope.$apply(function(){
                    angular.copy(nfcEvent.tag, tag);
                    // if necessary $state.go('some-route')
                });
            }, function () {
                console.log("Listening for NDEF Tags.");
            }, function (reason) {
                alert("Error adding NFC Listener " + reason);
            });

        });

        return {
            tag: tag,

            clearTag: function () {
                angular.copy({}, this.tag);
            }
        };
});
```

simonGleamorb, 2018-08-24:
Did you tried this plugin with version 0.7.3 ( phonegap-nfc@0.7.3 ) ?

don, 2018-10-09:
"nfc not defined" typically means the plugin is not installed correctly, or there is a problem with scoping. I haven't used Ionic1 for a long time, but I have a sample project you can try https://github.com/don/ionic-nfc-reader/tree/2b417dff9cd36b19a874170e66c257a513630e80.

closed 2018-10-09 by don