chariotsolutions/phonegap-nfc #347: Usage of "Choose an option" in Android

marioshtika, 2018-09-14:
In an Android phone if you get your NFC device close to your mobile. It prompts an action sheet that gives the user the option to choose which application he wants to open it with.

Can your plugin do this? Is there any thing I can do to make it happened?

Thank you in advanced.

simonGleamorb, 2018-10-09:
You could get your payload informations with this plugin and do what you want after. For example, display a modalWindow with your payloads info inside and some href or other. It depend what are the option and action that you want to use.

don, 2018-10-09:
@marioshtika When there are multiple apps that can handle NFC tags, Android will present the user with a list.

One strategy is to do something make your tags unique. You can use a mime type and restricting filtering to just your tags. This doesn't work as well with a generic app that scans multiple tags.

closed 2018-10-09 by don

marioshtika, 2018-10-09:
Hello @don and thank you for your response.
How can I use a mime type and restricting filtering to just my tags. 
Can this be done in a cordova project?
Can this be done through your plugin?
Or should this be done through the android project?

Thank you in advance.

don, 2018-10-09:
You can do this with Cordova, but you need to edit the [AndroidManifest.xml](Launching your Android Application when Scanning a Tag) to specify which tags launch your app. In your app, you need to use [nfc.addMimeTypeListener](https://github.com/chariotsolutions/phonegap-nfc#nfcaddmimetypelistener) to handle the tag that launched the app. The intent filter can filter on mime-type, plain-text, or a URL. See [the Android NFC docs](https://developer.android.com/guide/topics/connectivity/nfc/nfc#ndef-disc) for additional details. You'll need to figure out which NDEF type to use for your tags. You can make up a new mime type or use a url.