chariotsolutions/phonegap-nfc #349: nfc.share between Android and IOS

HerrZwerg, 2018-09-22:
Hi,

is there any sample for sending content between Android and IOS? I can read from IOS if the tag is emulated by another application on the device, but not from my Android application. 

So I guess my IOS version is working fine, so it must be something with my nfc.share method?

any help much appreciated
thx

ac3ju, 2018-10-02:
Hi ! Could you share your iOS code plz ? It doesn't work for me

HerrZwerg, 2018-10-02:
Hi ac3ju, can you describe you issue a little bit more in detail? Cannot send the full app as it includes some additional components as well...

ac3ju, 2018-10-02:
Just the part of NFC, what methods you are using and how plz. I see the "Ready to scan" but it doesn't scan :(

don, 2018-10-09:
Right now iOS can only read NFC tags. You might be able to do nfc.share() from Android and read with iOS but I haven't tried this yet. iOS NFC support is very limited. This plugin wraps the [CoreNFC](https://developer.apple.com/documentation/corenfc?language=objc) functionality.

HerrZwerg, 2018-10-09:
Hi, thank you for this info, but this is exactly what I try to archive. Send data from Android and read it on IOS. but this fails...

And if I use an ndef emulator it works perfectly fine...

don, 2018-10-09:
That's too bad it doesn't work. Unfortunately I can't do anything without iOS APIs. Maybe next year in iOS 13?

closed 2018-10-09 by don