chariotsolutions/phonegap-nfc #351: [iOS] Ready to scan, but nothing happen

ac3ju, 2018-10-03:
Hi,
I got the flyout "Ready to Scan" on my phone, but nothing is raised in my subscription.

`   try {
        this.nfc.beginSession().subscribe(() => {
          console.log('successfully beginSession');
          this.messages.push('successfully beginSession');
          this.nfc.addNdefListener(() => {
            console.log('successfully attached ndef listener');
            this.messages.push('successfully attached ndef listener');
            this.cekNFC();
          }, (err) => {
            console.log('error attaching ndef listener' + err);
            this.messages.push('error attaching ndef listener' + err);
          }).subscribe((event) => {
            console.log('received ndef message. the tag contains: ' + event);
            this.messages.push('received ndef message. the tag contains:' + event);

          });
        });
      } catch (error) {
        console.log('error ' + error);
      }
    });`

videmort, 2018-10-06:
Separat the NFC beginSession

function enableNFC() {
 nfc.beginSession();
}
When you init the NFC you will see one popOver on your iphone. 

Later add  your code -->

nfc.addNdefListener(() => {
            console.log('successfully attached ndef listener');
            this.messages.push('successfully attached ndef listener');
            this.cekNFC();
          }, (err) => {
            console.log('error attaching ndef listener' + err);
            this.messages.push('error attaching ndef listener' + err);
          }).subscribe((event) => {
            console.log('received ndef message. the tag contains: ' + event);
            this.messages.push('received ndef message. the tag contains:' + event);
          });

nfc is GLOBAL so is not on THIS context.... 

declare let nfc : any; at top of constructor

don, 2018-10-09:
Does your NFC tag contain an NDEF message? iOS only scans NDEF tags. 

You could try this sample project https://github.com/don/phonegap-nfc-ios

closed 2018-10-09 by don