chariotsolutions/phonegap-nfc #352: IOS, not return ID

videmort, 2018-10-06:
On iOS the plugin doesn't return ID of the NFC

{
    type: 'ndef',
    tag: {
        "isWritable": true,
        **"id": [4, 96, 117, 74, -17, 34, -128],**  NOT RETURN ON IOS (The NFC MAC)
        "techTypes": ["android.nfc.tech.IsoDep", "android.nfc.tech.NfcA", "android.nfc.tech.Ndef"],
        "type": "NFC Forum Type 4",
        "canMakeReadOnly": false,
        "maxSize": 2046,
        "ndefMessage": [{
            "id": [],
            "type": [116, 101, 120, 116, 47, 112, 103],
            "payload": [72, 101, 108, 108, 111, 32, 80, 104, 111, 110, 101, 71, 97, 112],
            "tnf": 2
        }]
    }
}

simonGleamorb, 2018-10-09:
There is a way to return UID with iOS.

For example, this user modified this original plugin and add some code to get TagId.
https://github.com/scottire/phonegap-nfc

The problem is if you want to use this forked plugin, when you will push your application on the appleStore, it will be rejected. Cause of some variables, security,.. I was facing this issue when i tried to use UID too.  

For my app, i used an unique ID inside my payload. (same as UID)

don, 2018-10-09:
@videmort the plugin can't include the UID until Apple provides a public API, otherwise like @simonGleamorb says, apple will reject that app.

closed 2018-10-09 by don