chariotsolutions/phonegap-nfc #355: App Crash at Apple Reviewers

anderssmedman, 2018-12-11:
This summer I created the first version of an iOS app that reads NDEF formatted tags flawless. This November I did som UI updates, test build on local iPhone 7 and Phone X with iOS 12.1, works perfect so I sent it to App Store Connect for review.

Apple review rejected, they give me only the crash log that says it crashes on startup, from the log:

```
Exception Type:  EXC_CRASH (SIGABRT)
Exception Codes: 0x0000000000000000, 0x0000000000000000
Exception Note:  EXC_CORPSE_NOTIFY
Termination Description: DYLD, Library not loaded: /System/Library/Frameworks/CoreNFC.framework/CoreNFC | Referenced from: /var/containers/Bundle/Application/1C2CF502-D05A-408B-AF51-ED10867F5FB4/twICEme.app/twICEme | Reason: image not found
Triggered by Thread:  0
```

Is this plugin supposed to work with iOS 12.1 ? Thanks!

halindrome, 2019-02-14:
I have this same issue now.  Does anyone have a work-around?   They removed the library from non-NFC devices in iOS 12.

don, 2019-02-14:
It would be helpful to know which calls are crashing the app. 

If it is crashing on startup, maybe it is the call checking if NFC is available during plugin initialization. Maybe there's a better way to do that with iOS 12?

It would help if someone could provide instructions on how to cause crash.

anderssmedman, 2019-02-17:
I do not know more than what I wrote in previous post. The app builds nicely and works as should in my dev env, it is Apple that says it crashes at startup. But, I dropped all iOS development and gave them away to an other company. No more Apple development because they change the gameplan over and over again.

mobidev111, 2019-02-22:
* problem: App got rejected by Apple Review
  - app crashes on start
Error: DYLD, Library not loaded: /System/Library/Frameworks/CoreNFC.framework/CoreNFC

* cause: CoreNFC is not always present, e.g. iPad w/iOS 12
   - CoreNFC not always present, e.g. iPad w/iOS 12, so must be weak linked
https://github.com/xamarin/xamarin-macios/commit/9f906918207f8ad19e08d82e0dc4cd2ed5924784
  - only (recent) devices with NFC have the framework (which might be new in iOS 12)
https://github.com/xamarin/xamarin-macios/issues/4628#issuecomment-413620630
   - since iOS 12
   - iOS12 does not include CoreNFC for older devices (iPhone 6s, SE and older).
  https://stackoverflow.com/questions/52562639/xcode-10-swift-4-app-with-corenfc-crashes-in-review-on-ios-12/52566753#52566753

* solution: mark CoreNFC as Optional
  - in Xcode: go to Project settings -> General -> Linked Frameworks and Libraries
  - Add CoreNFC.framework with status optional
https://stackoverflow.com/questions/52562639/xcode-10-swift-4-app-with-corenfc-crashes-in-review-on-ios-12/52566753#52566753
https://stackoverflow.com/questions/44946057/dyld-library-not-loaded-rpath-corenfc-framework-corenfc-ios11-and-xcode-9-be/52868674#52868674

* --> change CoreNFC linking in plugin.xml
  - most likely, still needs testing:
```
-        <framework src="CoreNFC.framework" />
+        <framework src="CoreNFC.framework" weak="true" />
```
https://github.com/chariotsolutions/phonegap-nfc/blob/master/plugin.xml#L137

Jahrenski, 2019-03-05:
Do you think it would be possible to automate this weak linking via cordova config.xml anyhow?

mobidev111, 2019-03-06:
As written (https://github.com/chariotsolutions/phonegap-nfc/issues/355#issuecomment-466500200): changing this plugin's [plugin.xml#L137](https://github.com/chariotsolutions/phonegap-nfc/blob/master/plugin.xml#L137) will automate this.

maxfahl, 2019-06-07:
I added the framework as optional to "Linked Frameworks and Libraries", but I still get the same error message:

```
dyld: Library not loaded: /System/Library/Frameworks/CoreNFC.framework/CoreNFC
  Referenced from: /private/var/containers/Bundle/Application/B97054D3-BF8B-4C7A-BFE2-631C1EF6B66C/App.app/Frameworks/CordovaPlugins.framework/CordovaPlugins
  Reason: image not found
(lldb) 
```

I have tried cleaning the build folder, no dice. Is there something else I'm missing here? I'm running 12.1 on iPhone SE.

AtanasChachev, 2019-09-25:
@maxfahl  I've managed to fix that by going into the xCode project -> Build Phases -> Link Binary With Libraries -> CoreNFC.framework should be `Optional` instead of `Required`. That's for xCode 11.

don, 2020-04-21:
#400 should fix this. Released in 1.1.1.

closed 2020-04-21 by don