chariotsolutions/phonegap-nfc #360: NFC reader doesn't work in Kiosk Mode after restart

KreoDev, 2019-01-23:
Hi,
When the phone is in kiosk mode everything works perfectly up until the phone has been restarted or shut down. When the phone has been restarted nfc doesn't scan up until you exit the app and then go back into it only does everything start to work again.
I'm having this issue with Android 7.0. Unfortunately all the phones I have access to have android 7.0. After doing some googling apparently Android 8.0 doesn't have the issue but unfortunately all my users are on Android 7.0.

I have created a demo ionic project that you can use to recreate the issue: https://github.com/KreoDev/Kiosk-NFC-Demo

To recreate the issue you just have to enable kiosk mode and then restart the phone and try to scan any NFC tag. Afterwards use the back button to exit the app and then go back into the app and then you will be able to start scanning NFC tags again.

fefoweb, 2019-01-29:
Same issue tested in Android 4.4.2 and Android 7.0
Using the go back button, the NFC return to scan tag.

looksystems, 2019-04-02:
Same issue here too with Android Oreo 8.1.

Seem to be related to this: 
https://stackoverflow.com/questions/41227623/android-nfc-intent-not-working-when-app-is-started-as-launcher

don, 2020-04-16:
This issue is really old. Closing. Generally these issues on specific devices and old versions of Android take a lot of time to track down. I don't have a good generic solution.

closed 2020-04-16 by don

carlo318, 2023-03-24:
However, I have the same issue with a tablet running on Android 10 and my hybrid ionic 1 app.
I will try to fix it with the [answer above](https://github.com/chariotsolutions/phonegap-nfc/issues/360#issuecomment-479171458).