chariotsolutions/phonegap-nfc #363: ERROR: Method 'channel:' not defined in Plugin 'NfcPlugin'

davidemorotti, 2019-04-16:
Hi,

I'm trying to develop an app with an NFC scanning functionality, here is my code:
```
    if(this.nfc.enabled()) {
      console.log('NFC enabled');
      this.nfc.beginSession((data) => {
        console.log(data)
      }, () => {
        console.log('error');
      })
    } else {
      console.log('NFC not enabled');
    }
```

When I try to run the app on my iOS device, this is the error I get:
`ERROR: Method 'channel:' not defined in Plugin 'NfcPlugin'`

Where is this "channel" method getting called?
Thanks for the help!

Davide

dawesc, 2019-04-30:
did you find a solution, i'm having the same error.

davidemorotti, 2019-04-30:
Nope, I developed the app using Swift 😂
I was able to do it faster than using Ionic packages. Sometimes the documentation is just not well written or present at all.

I’m sorry I can’t give you better news!

dawesc, 2019-04-30:
ok thanks for responding so quickly will keep my fingers crossed and use QR codes for now instead :D

bushev, 2019-07-05:
The same issue: `9.0.0 (cordova-lib@9.0.1)`

bushev, 2019-07-05:
I've fixed this error. Not sure, what the `channel` method should do exactly, probably it is needed for the plugin initialization. 

See https://github.com/bushev/device-com-nfc.cordova/commit/ea08e52a1c9395e387c5006fa51dcb6e24d15496

fixdaddy, 2019-11-28:
Thanks bushev! 

I have the same error with the latest IOS (10.15) and Xcode (11). I didn't see this correction (#363) in the 1.0.4 release notes. However, I see you committed this change on July 5. Shouldn't it be included in the 1.0.4 release?

don, 2020-04-16:
Fixed in #399

closed 2020-04-16 by don

coachdesign, 2020-07-20:
> Fixed in #399

During the test on the device, it gives me the same error.
I updated to the latest version available 1.2.0
```
PhoneGap NFC - Cordova Plugin
(c)2017 Don Coleman
ERROR: Method 'channel:' not defined in Plugin 'NfcPlugin'
FAILED pluginJSON = ["NfcPlugin866892554","NfcPlugin","channel",[]]
```
My `ionic info`
```
Ionic:

   Ionic CLI          : 6.10.1
   Ionic Framework    : ionic-angular 3.9.10
   @ionic/app-scripts : 3.2.3

Cordova:

   Cordova CLI       : 9.0.0 (cordova-lib@9.0.1)
   Cordova Platforms : android 9.0.0, ios 5.0.1
   Cordova Plugins   : cordova-plugin-ionic-webview 4.0.1, (and 18 other plugins)

Utility:

   cordova-res                          : not installed
   native-run (update available: 1.0.0) : 0.2.3

System:

   ios-deploy : 1.9.4
   ios-sim    : 8.0.2
   NodeJS     : v10.21.0
   npm        : 6.14.4
   OS         : macOS Catalina
   Xcode      : Xcode 11.5 Build version 11E608c
```