chariotsolutions/phonegap-nfc #366: [FEATURE REQUEST] Add support for non-NDEF tags for iOS 13

dpa99c, 2019-06-20:
iOS 13 (currently in Beta 2) opens up [non-NDEF NFC functionality](https://developer.apple.com/documentation/corenfc?language=objc) to 3rd party apps ([see here](https://techcrunch.com/2019/06/12/nfc-gets-a-lot-more-powerful-in-ios-13/)).

It would be great to add support for this to this plugin for the upcoming iOS 13 public release in September.

don, 2019-10-18:
What kind of non-NDEF tags do you want to read? Something like [the tag technology functions](https://github.com/chariotsolutions/phonegap-nfc#tag-technology-functions) or more like [host card emulation](https://github.com/don/cordova-plugin-hce)? I haven't studied the new docs yet so I don't know what's possible.

dpa99c, 2019-10-18:
For myself, supporting the [new APIs for MiFare tags](https://developer.apple.com/documentation/corenfc/nfcmifaretag?language=objc) would be ideal. If I had time, I'd have a go at implementing a PR for this but I'm currently snowed under with other OSS work.
Ideally the plugin would support all the new tag types supported by CoreNFC in the iOS 11 SDK. 

AFAIK the new CoreNFC operations don't open up enough low-level access to enable an HCE solution on iOS similar to what's possible on Android.

ReneDyhr, 2019-10-24:
Is there anything new to this? Would love to get access to tags with the NFCTagReaderSession instead of NFCNDEFReaderSession. NDEF doesn't support what I need.

lparcshinoda, 2020-10-24:
@don I see that now NFCTagReaderSession is supported with this plugin - is there a way to use something like "readMultipleBlocks" function from the returned interface (docs here https://developer.apple.com/documentation/corenfc/nfciso15693tag) ?