chariotsolutions/phonegap-nfc #367: Error compiling on build.phonegap

deadcalf, 2019-06-21:
When we try to compile the script on build.phonegap it shows to errors and stops compiling it says:
/var/gimlet/tmp/47026722182480/3614989/project/src/com/chariotsolutions/nfc/plugin/NfcPlugin.java:80: error: diamond operator is not supported in -source 1.6
    private final List<IntentFilter> intentFilters = new ArrayList<>();
                                                                   ^
  (use -source 7 or higher to enable diamond operator)
/var/gimlet/tmp/47026722182480/3614989/project/src/com/chariotsolutions/nfc/plugin/NfcPlugin.java:214: error: lambda expressions are not supported in -source 1.6
Any suggestions?

Adza93, 2020-01-26:
Same issue on ionic 3.9.8.

Any workarounds?

don, 2020-04-21:
PhoneGap build defaults to cli-6.5.0, which can't handle new Java or JavaScript. Update the phonegap-version to cli-9.0.0 by adding a preference to config.xml.

    <preference name="phonegap-version" value="cli-9.0.0" />
    <plugin name="phonegap-nfc" source="npm" />

closed 2020-04-21 by don