chariotsolutions/phonegap-nfc #369: Writing data to NFC tag on iOS 13

davidpadych, 2019-07-26:
Hi, are you thinking about adding the write functionality to the nfc tag for iOS 13? thanks

TimSchue, 2019-09-23:
Yes I would like to know too, thx

davidpadych, 2019-09-30:
Sample code: https://developer.apple.com/documentation/corenfc/creating_nfc_tags_from_your_iphone 

Sorry, i'm not a swift developer...

davidpadych, 2019-09-30:
fork is here: https://github.com/iotize-sas/device-com-nfc.cordova

don, 2019-10-18:
Writing NFC tags would be great. I'd like to do it eventually but I haven't looked into the details yet.

@davidpadych is that your repo? Can you send a pull request for the write functions?

davidpadych, 2019-10-18:
Sorry.. this is not my repo

gambarle, 2020-01-15:
I'm also very interested in being able to write to NFC tags with iOS 13 :-)

don, 2020-04-16:
Fixed in #399 and published in phonegap-nfc v1.1.0.

closed 2020-04-16 by don