chariotsolutions/phonegap-nfc #370: ERROR: Method 'registerTag:' not defined in Plugin 'NfcPlugin'

sumodevelopment, 2019-08-18:
I've been trying to get this working on iOS with no luck. I call the beginSession() function before attempting to start the scan. 

The console throws out the following error:

```
ERROR: Method 'registerTag:' not defined in Plugin 'NfcPlugin'
2019-08-18 19:20:27.952211+0200 Tickipay for promoters[1522:361428] FAILED pluginJSON = ["NfcPlugin26458284","NfcPlugin","registerTag",[]]
2019-08-18 19:20:36.289504+0200 Tickipay for
```

Here is a sample of the code: 
```
if (this.currentPlatform === 'ios') {
            await this.nfc.beginSession().toPromise();
            this.scanner$ = this.nfc.addNdefListener(() => {
                this.status = 'Scanner Ready...';
            }, () => {
                this.status = 'Failed to initialize scanner.';
            }).subscribe((e) => {
                this.tagDiscovered(e);
            })
        } else {
            this.scanner$ = this.nfc.addTagDiscoveredListener(() => {
                this.status = 'Scanner Ready...';
            }, () => {
                this.status = 'Failed to initialize scanner.';
            }).subscribe((e) => {
                this.tagDiscovered(e);
            });
        }
```

Any help would be greatly appreciated!

AtanasChachev, 2019-09-19:
I think that's because `addTagDiscoveredListener` is not supported on iOS , not pretty sure.. would be good if there's any solution / adding support for `addTagDiscoveredListener` for iOS.

don, 2019-10-18:
Also see https://github.com/don/phonegap-nfc-ios

don, 2020-04-16:
iOS works in v1.1.0. addTagDiscoveredListener is not supported on iOS.

closed 2020-04-16 by don