chariotsolutions/phonegap-nfc #371: Using with platform android version 8.x.x on API SDK 28

hamzahalvanaa, 2019-09-19:
I want to use NFC plugin for my Ionic 3 project using android version 8.1.0 on API SDK 28.

Because when uploading app into the Google Play Store require the latest API SDK 28, and the NFC plugin gives me the error when installing on android platform version 8.X.X.

It says Error: Unable to graft xml at selector "/manifest/uses-sdk" from "platforms\android\app\src\main\AndroidManifest.xml" during config install at ConfigFile_graft_child [as graft_child]

Please, someone, help me how to troubleshoot the plugin compatibility. Thank you.

sc0ttdav3y, 2019-09-20:
+1

I moved `<uses-sdk android:minSdkVersion="19" />` into the `<config-file target="AndroidManifest.xml" parent="/manifest">` block above and it works again.

biswas123, 2019-09-21:
I have the same problem, I am using phonegap build. Please have a fix for this on the plugin :)

mobidev111, 2019-09-22:
Could anyone of you open a pull request with this fix? then we can ping the maintainer to release a new version.

LuisEGR, 2019-09-24:
Fixed here: https://github.com/chariotsolutions/phonegap-nfc/pull/372

If you need to use this fix, I uploaded it to NPM so you can use it until the issue gets merged:

`npm i phonegap-nfc-fxmanifest`

or 

`ionic cordova plugin add phonegap-nfc-fxmanifest`

don, 2019-10-18:
#372 should fix this

closed 2019-10-18 by don