chariotsolutions/phonegap-nfc #373: supported ios devices

cyptus, 2019-10-01:
The doc says only "iPhone 7 and iPhone 7 Plus" is supported. Is this information up2date?
The iPad Pro models should have NFC capabilities too, the new iPad (2019) should also support NFC.

Does anyone has a complete list of supported Apple devices?

cyptus, 2019-10-09:
According to [ifixit iPad 7 Teardown](https://www.ifixit.com/Teardown/iPad+7+Teardown/126291#s243763) the iPad 2019 does have an [NXP 100VB27 NFC controller](https://www.techinsights.com/products/far-1810-801).
So it seems like the device would be supported by this plugin.

don, 2019-10-18:
That's an old comment. Most (all?) iOS devices after iPhone 7 should work. If an iOS device supports Apple Pay it has NFC.

closed 2019-10-18 by don