chariotsolutions/phonegap-nfc #375: Plugin not recognized as installed when testing with Cordova Ionic

samvanschaik, 2019-10-10:
After installing the plugin as explained in https://ionicframework.com/docs/native/nfc I am unable to get things working. I imagine it has something to do with the recent angular migration of the native ionic libraries to the /ngx sub-directory but am unable to find a solution. This is the warning I get when using the regular native library "'@ionic-native/nfc/"

`[ng] [console.warn]: "Install the NFC plugin: 'ionic cordova plugin add phonegap-nfc'"
[ng] [console.warn]: "Native: tried calling NFC.removeNdefListener, but the NFC plugin is not installed."
[ng] [console.warn]: "Install the NFC plugin: 'ionic cordova plugin add phonegap-nfc'"
[ng] [console.log]: "Angular is running in the development mode. Call enableProdMode() to enable the production mode."
[ng] [console.log]: "calling _sync"
[ng] [console.error]: "ERROR" "plugin_not_installed"
[ng] [console.info]: "[WDS] Live Reloading enabled."
[ng] [console.log]: "calling _reload"
[ng] [console.log]: "done _reloading"
[ng] [console.log]: "Ionic Native: deviceready event fired after 836 ms"`

If I specify the Angular version and import the /ngx subdirectory I get a different error entirely. 

`Uncaught TypeError: Object(...) is not a function
    at index.js:405
    at Module../node_modules/@ionic-native/nfc/index.js (index.js:599)
    at __webpack_require__ (bootstrap:84)
    at Module../src/app/app.module.ts (app.component.ts:12)
    at __webpack_require__ (bootstrap:84)
    at Module../src/main.ts (main.ts:1)
    at __webpack_require__ (bootstrap:84)
    at Object.0 (main.ts:12)
    at __webpack_require__ (bootstrap:84)
    at checkDeferredModules (bootstrap:45)`

This error comes up the moment the app is compiled (with 'ionic serve --devapp'), unlike the plugin not installed error, that only appears when a method from the plugin is used. 

The plugin does work correctly on PhoneGap itself, but not on ionic.

don, 2019-10-18:
> The plugin does work correctly on PhoneGap itself, but not on ionic.
This sounds like a problem with Ionic or the Ionic wrapper.

samvanschaik, 2019-10-18:
> > The plugin does work correctly on PhoneGap itself, but not on ionic.
> 
> This sounds like a problem with Ionic or the Ionic wrapper.
> Maybe the manifest change in #372 helps?

It took me a while but it turns out that there were multiple issues at hand. It looks like Ionic Cordova past version 8 doesn't work with the plug-in. It also seems like Ionic Cordova 8 does not recognize Java Versions beyond 9 as being greater than 1. The Ionic Devapp also does not support the NFC plug-in, which makes debugging an absolute drag.

I might have some of these details wrong,or remember them incorrectly, but I am running Cordova 8, Ionic 4 with Java 8 and am able to use the NFC plug-in when I build with "ionic cordova run android", at least.

don, 2019-10-18:
@samvanschaik This is helpful info. I'll see if I can get it working on the latest of everything.

samvanschaik, 2019-10-19:
Best of luck! @don

don, 2019-10-22:
I can duplicate the problem on PhoneGap Build. I'm hoping there's a way to set the language level to 1.8, but I haven't found it yet.

samvanschaik, 2019-10-23:
Do you reckon it's possible to make the NFC plugin work on the Ionic Devapp as well or is this out of your reach? Having to build an app to test with the plugin makes development quite rough, but I suppose you are not a user of the Ionic framework yourself and that Phonegap doesn't have this issue, or?

andersborgabiro, 2019-10-24:
Maybe related: I get NO_NFC on an iPhone 7 with iOS 13.1.13. Do certain i7s not support NFC?

samvanschaik, 2019-10-29:
I believe that XML requirement is just for iPhone. The ID requirement is also iPhone related. The specifics escape me but iPhones can't properly read out all of an NFC in the same way an Android can. On the newer models, at least. Pretty sure you can find more about this online. I heard they were considering removing this in the future, however. It's some kind of safety measure.

andersborgabiro, 2019-10-29:
@samvanschaik Yup, iOS only. To me this is nothing extraordinary, but I wasn't aware of it until I read a general description about using NFC on iOS. This has nothing to do with any limitations in the implementation per se.

For both Android and iOS I only use the NDEF listener (not locking to certain MIME types), so there might be differences when using MIME. Instead I support any tag with Text och URL, and the user needs to know what to apply. It's a rather generic implementation for now (input to forms, so any text will do). Task-specific implementations will surely be more discriminating.

closed 2020-04-16 by don

amolht, 2020-07-06:
How to test nfc ans hce virtually on machine without using real devices like POS machine or nfc enabled Device