chariotsolutions/phonegap-nfc #378: Error during cordova platform install android 8.1.0

studiostart, 2019-10-25:
Hello,
after installing the latest versione of this NFC plugin, if I launch:

cordova platform add android@8.0.0

the installation process is completed and everything works ok. Instead if I launch:

cordova platform add android@8.1.0 or android@latest

I have this error (C:\L\intelliclima-app\ is my Cordova app):
Failed to install 'phonegap-nfc': Error: Unable to graft xml at selector "/manifest/uses-sdk" from "C:\L\intelliclima-app\platforms\android\app\src\main\AndroidManifest.xml" during config install
    at ConfigFile_graft_child [as graft_child] (C:\L\intelliclima-app\node_modules\cordova-common\src\ConfigChanges\ConfigFile.js:122:19)
    at PlatformMunger_apply_file_munge [as apply_file_munge] (C:\L\intelliclima-app\node_modules\cordova-common\src\ConfigChanges\ConfigChanges.js:81:34)
    at munge_helper (C:\L\intelliclima-app\node_modules\cordova-common\src\ConfigChanges\ConfigChanges.js:252:14)
    at PlatformMunger.add_plugin_changes (C:\L\intelliclima-app\node_modules\cordova-common\src\ConfigChanges\ConfigChanges.js:159:12)
    at C:\L\intelliclima-app\node_modules\cordova-common\src\PluginManager.js:123:29
    at _fulfilled (C:\L\intelliclima-app\node_modules\q\q.js:854:54)
    at C:\L\intelliclima-app\node_modules\q\q.js:883:30
    at Promise.promise.promiseDispatch (C:\L\intelliclima-app\node_modules\q\q.js:816:13)
    at C:\L\intelliclima-app\node_modules\q\q.js:877:14
    at runSingle (C:\L\intelliclima-app\node_modules\q\q.js:137:13)
Unable to graft xml at selector "/manifest/uses-sdk" from "C:\L\intelliclima-app\platforms\android\app\src\main\AndroidManifest.xml" during config install

Any idea?

zilles-edv, 2019-11-23:
I have the same problem. Did you find a solution?

benediktcarda, 2020-01-26:
Same problem here.

Jahrenski, 2020-03-02:
Same here! Need to build twice for it to go away

don, 2020-04-16:
I think this is fixed now by some other updates. I've been building with cordova-android@8.1.0 with Cordova CLI 9.0.0 on macOS all week without problems. Re-open this issue or create a new one if it's still a problem.

closed 2020-04-16 by don