chariotsolutions/phonegap-nfc #380: UIDs with iOS13 and Ionic Cordova

JNBG, 2019-10-28:
Hi,
tried to create an ios-ionic-cordova app to simply read out an UID from an NFC Chip using this Code:

```
iosNfcButton.addEventListener('click', function() {
  thisNFC.beginSession((data) => {},
    (err) => {}
      ).subscribe((event) => {
        thisNFC.addNdefListener(() => {
          console.log('connected ndef listener?!')
        }, (err) => {
          console.log('error attaching ndef listener', err);
        }).subscribe((event) => {
         alert(JSON.stringify(event))
        }
      });
    })
  })
});
```
Baiscally just alert anything from the event in JSON so it's from so I can read it without a console. What I got back was:
```
{
  "isTrusted":false,
    "tag":{
      "ndefMessage":[{
        "tnf": 1,
        "id": [],
        "payload": [33,23,23,2,3],
        "type": [85]
     }]	
  }
}
```
As you can see the ID is empty. Is there any way I can access this now or maybe in future releases?

vuptt, 2019-11-07:
@don I just created a forked repo to return the uid, can you check it out and add this feature. Thanks

[https://github.com/vuptt/phonegap-nfc](https://github.com/vuptt/phonegap-nfc)

GeorgeD19, 2019-11-11:
Great work @vuptt however your fork does pull the IDs however when using addTagDiscoveredListener() and scanning an NXP MIFARE Classic 4k (ISO 14443-3A) it no longer is detected.

vuptt, 2019-11-13:
@GeorgeD19 I don't think addTagDiscoveredListener work on ios.

GeorgeD19, 2019-11-13:
@vuptt I can confirm addTagDiscoveredListener does work on iOS however with the new implementation in your fork it misses out specifically MIFARE classic, it's fine with other tags like MIFARE ultralight. I do not think this any bug though, when I read into it I think apple outright doesn't like MIFARE classic when it comes to grabbing the uid. You can scan MIFARE classic without the uid no problem.

Patrick-89, 2019-12-06:
Hello together. Thank you all for your great work. We are really happy to use your work daily.

But we need to read out the UID of the NFC Chips. But this is now not possible.
Do you have any Idea how this app had solved the issue about scanning out the UID?
https://apps.apple.com/de/app/nfc-tools/id1252962749

Or what is missing in the Cordova plugin in order to read out the NFC UID?

Thank you.

don, 2020-04-16:
Fixed by #399. Released in phonegap-nfc 1.1.0. You can get the UID on some tags.

See [nfc.scanTag](https://github.com/chariotsolutions/phonegap-nfc#nfcscantag)

closed 2020-04-16 by don