chariotsolutions/phonegap-nfc #384: phonegap-nfc plugin causes error on startup on IOS

alexdev888, 2019-11-12:
Platform: Ionic + Capacitor + Angular
Target device: iPhone SE

When I run my app on iPhone after I install phonegap-nfc plugin the app crashes.
The idea is that iPhone SE has NFC only for payments. 
I need at list app to start without error, but after installing NFC plugin I get this on compiling App in Xcode: 

dyld`__abort_with_payload:
    0x100902390 <+0>:  mov    x16, #0x209
    0x100902394 <+4>:  svc    #0x80
->  0x100902398 <+8>:  b.lo   0x1009023b0               ; <+32>
    0x10090239c <+12>: stp    x29, x30, [sp, #-0x10]!
    0x1009023a0 <+16>: mov    x29, sp
    0x1009023a4 <+20>: bl     0x1009017d0               ; cerror_nocancel
    0x1009023a8 <+24>: mov    sp, x29
    0x1009023ac <+28>: ldp    x29, x30, [sp], #0x10
    0x1009023b0 <+32>: ret    

How can I make this plugin to not be loaded if phone does not support NFC on IOS?
Any solution?

JoshuvaGeorge03, 2020-02-25:
I too have this issue any advice, please?

JoshuvaGeorge03, 2020-02-25:
@alexdev888 How do you fixed this issue?

don, 2020-04-09:
The fix should be to make a weak reference to CoreNFC in plugin.xml. I haven't duplicated this yet, but pull request #395 should be the fix.

JoshuvaGeorge03, 2020-04-10:
@don how to use that pull request?

don, 2020-04-10:
To try a different branch, remove the existing plugin

    cordova plugin rm phonegap-nfc

Then install the version from the PR using the git URL

    cordova plugin add https://github.com/M3skar/phonegap-nfc

I managed to track down an iPhone SE so hopefully I'll be able to test this soon.

don, 2020-04-10:
I can not duplicate this.

- iPhone SE with iOS 13.2
- Cordova 9.0.0
- cordova-ios 5.1.1
- Xcode 11.4 (11E146)

If someone can duplicate this, please post details here.

just4give, 2020-04-10:
I am facing same issue during start up ...  

My platform details 
```
Ionic:

   Ionic CLI                     : 6.2.2 (/usr/local/lib/node_modules/@ionic/cli)
   Ionic Framework               : @ionic/angular 5.0.7
   @angular-devkit/build-angular : 0.803.26
   @angular-devkit/schematics    : 8.3.26
   @angular/cli                  : 8.3.26
   @ionic/angular-toolkit        : 2.2.0

Capacitor:

   Capacitor CLI   : 2.0.0
   @capacitor/core : 2.0.0

Utility:

   cordova-res (update available: 0.11.0) : 0.10.0
   native-run (update available: 1.0.0)   : 0.3.0

System:

   NodeJS : v10.13.0 (/usr/local/bin/node)
   npm    : 6.13.2
   OS     : macOS Mojave

XCode 11.2
iOs 13 on iPhone 11
```

Log 

```
To Native Cordova ->  NfcPlugin channel NfcPlugin433258371 ["options": []]
2020-04-10 15:53:53.159112-0400 App[29262:18302678] PhoneGap NFC - Cordova Plugin
2020-04-10 15:53:53.159143-0400 App[29262:18302678] (c)2017 Don Coleman
⚡️  [log] - onscript loading complete
2020-04-10 15:53:53.261177-0400 App[29262:18302678] [CoreNFC] 00000002 824881a0 -[NFCHardwareManager areFeaturesSupported:outError:]:156  XPC Error: Error Domain=NSCocoaErrorDomain Code=4099 "The connection to service on pid 0 named com.apple.nfcd.service.corenfc was invalidated." UserInfo={NSDebugDescription=The connection to service on pid 0 named com.apple.nfcd.service.corenfc was invalidated.}
2020-04-10 15:53:53.462803-0400 App[29262:18302678] [CoreNFC] 00000002 824881a0 -[NFCHardwareManager areFeaturesSupported:outError:]:156  XPC Error: Error Domain=NSCocoaErrorDomain Code=4099 "The connection to service on pid 0 named com.apple.nfcd.service.corenfc was invalidated from this process." UserInfo={NSDebugDescription=The connection to service on pid 0 named com.apple.nfcd.service.corenfc was invalidated from this process.}
2020-04-10 15:53:53.765281-0400 App[29262:18302678] [CoreNFC] 00000002 824881a0 -[NFCHardwareManager areFeaturesSupported:outError:]:156  XPC Error: Error Domain=NSCocoaErrorDomain Code=4099 "The connection to service on pid 0 named com.apple.nfcd.service.corenfc was invalidated from this process." UserInfo={NSDebugDescription=The connection to service on pid 0 named com.apple.nfcd.service.corenfc was invalidated from this process.}
2020-04-10 15:53:54.167782-0400 App[29262:18302678] [CoreNFC] 00000002 824881a0 -[NFCHardwareManager areFeaturesSupported:outError:]:156  XPC Error: Error Domain=NSCocoaErrorDomain Code=4099 "The connection to service on pid 0 named com.apple.nfcd.service.corenfc was invalidated from this process." UserInfo={NSDebugDescription=The connection to service on pid 0 named com.apple.nfcd.service.corenfc was invalidated from this process.}
2020-04-10 15:53:54.669139-0400 App[29262:18302678] NFC Support is NOT available
Error: Plugin NfcPlugin does not respond to method call channel:.
Ensure plugin method exists and uses @objc in its declaration
⚡️  WebView loaded
```

Any help is highly appreciated !

JoshuvaGeorge03, 2020-04-11:
@don  is this pull request support WKWebView? 
 
Is this have every feature as current nfc plugin?

don, 2020-04-11:
@just4give you're seeing a different error maybe related to #363

just4give, 2020-04-11:
@don thanks for sharing that. I installed that PR and now enabled() is working but beginSession does not seem to work. Not getting enough information in the log. Not sure if you can put some light on this?

```
2020-04-11 17:18:14.982306-0400 NFC Scanner[30488:18767392] PhoneGap NFC - Cordova Plugin
2020-04-11 17:18:14.982333-0400 NFC Scanner[30488:18767392] (c)2017 Don Coleman
2020-04-11 17:18:14.985133-0400 NFC Scanner[30488:18767392] channel :-)
2020-04-11 17:18:14.985218-0400 NFC Scanner[30488:18767392] [Process] kill() returned unexpected error 1

2020-04-11 17:22:55.057910-0400 NFC Scanner[30488:18767392] lets start NFC session
2020-04-11 17:22:55.058441-0400 NFC Scanner[30488:18767392] [Process] kill() returned unexpected error 1
```

Code

```
ionViewDidEnter() {
    let that = this;

    this.nfc.enabled().then(()=>{
      alert('NFC enabled');
      console.log("lets start NFC session ")
      this.nfc.beginSession(
        success => {
          alert('Session started');
        },
        error => alert('Failed to start session ')
      );

    
    },
    (error)=>{alert('NFC Not enabled'+error)});

  }

```

As error message does not indicate anything clearly , I am kind of lost where to go from here :(

don, 2020-04-16:
Fixed in #399 

beginSession is deprecated. Use scanNdef instead since it returns a Promise. The Ionic wrapper isn't updated yet, so use `(<any>window).nfc.scanNdef()` until it's updated.

closed 2020-04-16 by don