chariotsolutions/phonegap-nfc #393: Is there anyway to detect a canceled scan?

joshstrange, 2020-02-03:
I was showing something in the background when you started a scan but if you press the iOS "Cancel" button then you get stuck (in my app). Is there a way to detect when a user does that or do I need to find another solution?

fixdaddy, 2020-02-09:
I have the same question. IOS surely allows it :https://developer.apple.com/documentation/corenfc/building_an_nfc_tag-reader_app.

I don't feel competent to tackle this update myself, though.

fixdaddy, 2020-02-09:
Looking at the file phonegap-nfc/blob/master/src/ios/NfcPlugin.m, I feel there is code to deal with this (lines 86-93). You would have to look for an error that reads "readerSessionInvalidationErrorUserCanceled". 
Logically, your code should be looking for it in the failure callback of the nfc.beginSession method. Unfortunately, I don't see clear evidence that there is code for a failure callback (lines 33-39).
Help from experts would be greatly appreciated here... 
Thanks in advance!

fixdaddy, 2020-02-11:
I have tested with an iPhone and can confirm the following:
   - The success callback is called when the IOS is ready to scan a NFC tag.
   - The failure callback of nfc.beginSession is not called when the session is invalidated.
   - Line 87 of NfcPlugin.m outputs as log the following:
      - When the user cancels the scan: "didInvalidateWithError Session invalidated by user (null)"
      - When a scan is performed : "didInvalidateWithError First NDEF tag read (null)"

Line 87 reads as follows: NSLog(@"didInvalidateWithError %@ %@", error.localizedDescription, error.localizedFailureReason);

From the log output, it seems that the error.localizedDescription has a value that could be used in the code using the plugin to distinguish between actions to take when the user canceled the scan.

@don on you mentioned in the Readme file that if someone had a use case and idea how to implement, a ticket for discussion should be opened. I believe all conditions are met.

I suggest to use the failure callback of the nfc.beginSession to pass the localizedDescription. 

I volounteer to test an improved version of the NfcPlugin.m file.

Thanks in advance!

wreiske, 2020-02-17:
Add me on that list as a volunteer to test! +1

cytostatic, 2020-03-09:
I have the same Problem. 
Is there any solution yet?

It is a cool plugin, but with this it is unusable for my App.

fixdaddy, 2020-03-09:
Hello,

   No solution yet, unfortunately. I agree the plugin is cool. 

   I am still hopeful someone who knows the language and architecture of the plugin/Cordova will step in.

don, 2020-04-10:
The failure callback for beginSession is the right place to send this. I'm working on some other iOS fixes and will try to add this too.

don, 2020-04-10:
Scan cancelled error is never being sent because the callback isn't being kept.

https://github.com/chariotsolutions/phonegap-nfc/blob/df3afc165d5546bccbedca66ec1496bd04b36e6d/src/ios/NfcPlugin.m#L98-L101

Better behavior would be to fire the success callback when it starts, then the error if there is an error. This is fine for callbacks, but doesn't work promises. There may be breaking change in the next release.

Does anyone care that the session started callback?

fixdaddy, 2020-04-10:
> The failure callback for beginSession is the right place to send this. I'm working on some other iOS fixes and will try to add this too.

That's great news Don! I see that you self-assigned #363. Awesome. I am using the fix provided.

fixdaddy, 2020-04-10:
> Scan cancelled error is never being sent because the callback isn't being kept. [...]

I'm sorry, Don! You lost me with this question. If it does what you wrote above "Better behavior would be to fire the success callback when it starts, then the error if there is an error.", I'm all for it.

don, 2020-04-11:
@fixdaddy the next release will call the error callback when the user cancels the scan

fixdaddy, 2020-04-12:
> @fixdaddy the next release will call the error callback when the user cancels the scan

And when the scan times out, I guess? I think IOS reacts the same way in both cases.

don, 2020-04-16:
The Ionic wrapper for nfc.beginSession doesn't work. beginSession is deprecated and replaced with scanNdef which returns a Promise. This is published in phonegap-nfc-1.1.0. 

See example code in https://github.com/chariotsolutions/phonegap-nfc#nfcscanndef

Ionic wrapper will be patched soon, in the meantime you can

closed 2020-04-16 by don