chariotsolutions/phonegap-nfc #398: UIWebView Deprecation warning by Apple

JoshuvaGeorge03, 2020-03-26:
Thanks for the wonderful plugin.

Please refer below. According to apple, When the app contains a reference to UIWebView, they won't allow permission into AppStore. I have found out phonegap-nfc has a reference to UIWebView. Please advise, what to do to fix this issue?

> If your app still embeds web content using the deprecated UIWebView API, we strongly encourage you to update to WKWebView as soon as possible for improved security and reliability. WKWebView ensures that compromised web content doesn’t affect the rest of an app by limiting web processing to the app’s web view. And it’s supported in iOS and macOS, and by Mac Catalyst.

The App Store will no longer accept new apps using UIWebView as of April 2020 and app updates using UIWebView as of December 2020.

(https://developer.apple.com/news/?id=12232019b)

tzeroem, 2020-04-02:
This plugin supports WKWebView.

don, 2020-04-08:
@JoshuvaGeorge03 your app got rejected because of this?  It's a conditional reference that can probably be removed like @tzeroem suggests.

JoshuvaGeorge03, 2020-04-09:
@don Could you please advise how to use WKWebView in PhoneGap-NFC?

JoshuvaGeorge03, 2020-04-09:
@tzeroem Could you please say how to remove UIWebView instance from NfcPlugin.m? Please advice.

don, 2020-04-09:
@JoshuvaGeorge03 newer versions of Cordova/PhoneGap should use WKWebView

The ObjectiveC code in the plugin needs to execute some Javascript in the UI. This code checks to see if your app is using UIWebView or WKWebView and calls the function on the correct one.

https://github.com/chariotsolutions/phonegap-nfc/blob/df3afc165d5546bccbedca66ec1496bd04b36e6d/src/ios/NfcPlugin.m#L117-L122

I don't want to remove that code immediately because it will break older apps. If your app was rejected by Apple because of this UIWebView let me know and I'll bump up the priority on this fix.

JoshuvaGeorge03, 2020-04-10:
@don I have got a warning message from apple for now.  But it should be done for every app I think. Because apple will stop accepting apps with UIWebView reference from December 2020.

Please check this link for more info

https://developer.apple.com/news/?id=12232019b

don, 2020-04-11:
@JoshuvaGeorge03 this is just a warning. Ignore it for now. The next release will fix it.

JoshuvaGeorge03, 2020-04-12:
@don Thanks, keenly waiting for that

don, 2020-04-16:
PR #399 removed UIWebView. Changes published in phonegap-nfc-1.1.0.

closed 2020-04-16 by don