chariotsolutions/phonegap-nfc #402: iOS - Dismiss tag reading dialogue as quick as possible

chrispahm, 2020-04-20:
First off, thanks for the amazing plugin @don !
I've been tinkering with the iOS 13 tag reading/writing capabilities of this plugin. One thing I noticed is that when reading or writing a tag, one would have to wait for the default iOS animation (with the blue checkmark) to be finished before being able to use the app again. There also doesn't seem to be a way to dismiss it, so I think it results in a bad user experience:
Here's a GIF that illustrates the issue (not mine):

[image]

I found an article discussing the issue, and posting possible solutions in Swift (this is where the GIF is from): https://itnext.io/swift-corenfc-as-quick-as-possible-79544796cd0a
Unfortunately, I do not have any experience with Objective-C and Swift so I was having trouble doing it myself and posting a PR. I would appreciate if you could have a look though!

Here's a GIF from the above article of how their proposed solution looks like in action:
[image]

don, 2020-04-21:
@chrispahm that's an interesting article. Currently, you can cancel the scan and dismiss the dialog from JavaScript with [nfc.cancelScan](https://github.com/chariotsolutions/phonegap-nfc#nfccancelscan). The scanNfdef and scanTag APIs invalidate the NFCReader session automatically after a scan or error. The API calls return promises and are set up for scanning one tag at a time. This is inferior to the Android behavior, but works good enough. The plugin can be expanded later when we need different behavior for an app or someone hires us to make enhancements.

guizmo51, 2021-06-16:
Hi @don  thnaks too for this amazing plugin (i'm working on a fork using web nfc for Capacitor V3 :) ) 

About the cancelScan method ... when I use it the iOS dialog isn't immediately removed (2-3 seconds delay or more) do you know if something can improve it ? 

Also, is it possible to add a custom listener on the "dialog removed" event ? (readerSession ended) 

Thanks !