chariotsolutions/phonegap-nfc #404: Error write Android

zigou13, 2020-04-28:
IONIC 5
Hello, when writing NFC, I got this error

Attempt to invoke virtual method 'android.os.Parcelable android.content.Intent.getParcelableExtra (java.lang.String)' on a null object reference

For reading the NFC, it works perfectly, here is the code below

`readNFC() {
    console.log('read');
    this.nfc.addNdefListener((addndef) => {
      console.log('17 : ', addndef);
      //this.presentAlert('ok');
    }, (err) => {
      console.log('20 : ', err);
      this.presentAlert('ko' + err);
    }).subscribe((event) => {
      console.log('Tag : ', event);
      //console.log(JSON.stringify(event));
    });
  }

  writeNFC() {
    console.log('write');
    const message = this.ndef.textRecord('Hello world');
    this.nfc.write([message]).then(data => {
      console.log(data);
      console.log('Write successfully');
    }
    ).catch(err => {
      console.log(err);
      console.log('Write KO !');
    });
  }`

zigou13, 2020-04-30:
@don please

don, 2020-06-18:
The call to nfc.write needs to happen inside the NDEF listener callback.

zigou13, 2020-06-18:
ok thank you

closed 2020-06-18 by zigou13

don, 2020-06-18:
Here's an old example that should still work. I add a tag discovered listener, passing my `writeTag` function as the callback https://github.com/don/phonegap-nfc-writer/blob/9f32ad17115457f75319b41c890ba1669a2b18a5/www/main.js#L32

Then in `writeTag`, I ignore any data in the nfcEvent and write my new data to the tag.

PhamHuyHiep45200, 2024-08-09:
@don please help me. I am implementing NFC P2P using nfc.share(), but it always returns NDEF_PUSH_DISABLED even though I have enabled NFC on the device and checked that there is no Android Beam available. Below is my code
nfc.share(
      [ndef.textRecord("Hello")],
      () => {
        alert("Success");
      },
      (err) => alert(JSON.stringify(err))
    );