chariotsolutions/phonegap-nfc #406: Error whille PhoneGap Build

mephomat, 2020-05-07:
Hi, 

since the latest Version I get an error while building with PhoneGap Build Service.

[image]

Here is the log:

The following build commands failed:
	CompileC /Library/Developer/Xcode/DerivedData/HBR_DEMO_CLIENT-cdnedeskkhbacpektbrmxeziaeqg/Build/Intermediates.noindex/ArchiveIntermediates/HBR_DEMO_CLIENT/IntermediateBuildFilesPath/HBR_DEMO_CLIENT.build/Release-iphoneos/HBR_DEMO_CLIENT.build/Objects-normal/armv7/NfcPlugin.o HBR_DEMO_CLIENT/Plugins/phonegap-nfc/NfcPlugin.m normal armv7 objective-c com.apple.compilers.llvm.clang.1_0.compiler
(1 failure)
Error: xcodebuild: Command failed with exit code 65
    at ChildProcess.whenDone (/private/project/cordova/node_modules/cordova-common/src/superspawn.js:135:23)
    at ChildProcess.emit (events.js:182:13)
    at maybeClose (internal/child_process.js:962:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)

Is this a Problem on my side or a bug?

Thanks

PVArgosDN, 2020-05-18:
Hi there
i have the same issue
i use client cli-9.0.0 and latest version of plugin
i cannot make a build using previous versions too...
thanks for help
Philippe

The following build commands failed:
	CompileC /Library/Developer/Xcode/DerivedData/ECD_TESTS-eeogysjvpqcbyfeolallnclkoscq/Build/Intermediates.noindex/ArchiveIntermediates/ECD_TESTS/IntermediateBuildFilesPath/ECD_TESTS.build/Release-iphoneos/ECD_TESTS.build/Objects-normal/armv7/NfcPlugin.o ECD_TESTS/Plugins/phonegap-nfc/NfcPlugin.m normal armv7 objective-c com.apple.compilers.llvm.clang.1_0.compiler
(1 failure)
Error: xcodebuild: Command failed with exit code 65
    at ChildProcess.whenDone (/private/project/cordova/node_modules/cordova-common/src/superspawn.js:135:23)
    at ChildProcess.emit (events.js:182:13)
    at maybeClose (internal/child_process.js:962:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)

don, 2020-06-18:
Is this a duplicate of #367? I tested PhoneGap Build with Android and it worked OK. I didn't test with iOS. I don't currently have any valid keys uploaded to PGB.  If you can work with the PhoneGap Build team to track down the cause, update info here or submit a patch.

johnsontsoi, 2020-06-27:
> Hi there
> i have the same issue
> i use client cli-9.0.0 and latest version of plugin
> i cannot make a build using previous versions too...
> thanks for help
> Philippe
>
> The following build commands failed: [...]

Same here. I just removed the plugin installed in several months ago, and added it back with the latest version, then it prompts this error when building the iOS platform; No build error for Android.

PVArgosDN, 2020-08-06:
@don ,
I have inspected the log file once the error appeared.
Please find below what I found 

```
/Library/Developer/Xcode/DerivedData/ECD_NLP_ARGOS-fndxncbykircieavjqomfwafhohb/Build/Intermediates.noindex/ArchiveIntermediates/ECD_NLP_ARGOS/IntermediateBuildFilesPath/ECD_NLP_ARGOS.build/Release-iphoneos/ECD_NLP_ARGOS.build/Objects-normal/armv7/NfcPlugin.dia -c /project/ECD_NLP_ARGOS/Plugins/phonegap-nfc/NfcPlugin.m -o /Library/Developer/Xcode/DerivedData/ECD_NLP_ARGOS-fndxncbykircieavjqomfwafhohb/Build/Intermediates.noindex/ArchiveIntermediates/ECD_NLP_ARGOS/IntermediateBuildFilesPath/ECD_NLP_ARGOS.build/Release-iphoneos/ECD_NLP_ARGOS.build/Objects-normal/armv7/NfcPlugin.o
In file included from /project/ECD_NLP_ARGOS/Plugins/phonegap-nfc/NfcPlugin.m:7:
/project/ECD_NLP_ARGOS/Plugins/phonegap-nfc/NfcPlugin.h:14:65: error: cannot find protocol declaration for 'NFCTagReaderSessionDelegate'; did you mean 'NFCReaderSessionDelegate'?
@interface NfcPlugin : CDVPlugin <NFCNDEFReaderSessionDelegate, NFCTagReaderSessionDelegate> {
                                                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                                NFCReaderSessionDelegate
In module 'CoreNFC' imported from /project/ECD_NLP_ARGOS/Plugins/phonegap-nfc/NfcPlugin.h:11:
```
it seems that there is an error in the name of a variable **NFCTagReaderSessionDelegate**

it is not a duplicate from #367 

Thanks for your help
Philippe

PVArgosDN, 2021-01-18:
Now i am not using phonegap anymore to build my app this plugin works perfectly