chariotsolutions/phonegap-nfc #407: Error System resource unavailable when attempting to write nfc tag on iOS

TheLazyHatGuy, 2020-05-13:
I try to use the following code to write to an NFC tag on iOS
```
this.nfc.write(tagMessage).then(() => {
            this.readingTag = false;
            this.getLocation(tagUUID);
          }).catch((e) => {
            console.log('Then.Catch()');
            this.updateLoading(1);
            this.statusMessage = 'ERR: ' + e;
            this.readingTag = false;
            console.error('Write to tag error', e);
            this.nfcError(e);
            });
```

But every time I do I get the error `[error] - ERROR Error: Uncaught (in promise): System resource unavailable\`

TheLazyHatGuy, 2020-05-13:
I'm using an iPhone 7 on iOS 13 and ionic-native/NFC plugin 5.25.0 phonegap-nfc 1.1.1

don, 2020-06-18:
This works on my iPhone 7

        var message = [
            ndef.textRecord("Hello phonegap-nfc")
        ];

        nfc.write(
            message,
            success => console.log("Wrote Message"),
            error => console.log(error)
       );

don, 2020-07-09:
Can't duplicate

closed 2020-07-09 by don