chariotsolutions/phonegap-nfc #409: makeReadOnly for iOS

amenders, 2020-05-18:
It looks like makeReadOnly is only supported on Android - can this functionality be brought to iOS as well?
https://developer.apple.com/documentation/corenfc/nfcndeftag/3075573-writelockwithcompletionhandler?language=objc

don, 2020-06-19:
It looks like that iOS API could work. The trick is figuring out how to make it work with the javascript API. nfc.makeReadOnly could start a new scan session just for locking tags. Locking the tag immediately after writing would probably be preferable, but is tougher to implement because the app code loses control once the scan session starts.

PVArgosDN, 2021-01-18:
Dear Don,

would it be possible to add this feature for ios ? this would let me finish my project and make it works on ios as on android platform
thank you for your help
Philippe

chautnguyen4392, 2024-07-16:
It's really useful if we can add this feature for iOS. I tried using `makeReadOnly` on iOS but it didn't work
```
2024-07-16 11:01:01.003219+0700 ERROR: Method 'makeReadOnly:' not defined in Plugin 'NfcPlugin'
2024-07-16 11:01:01.003296+0700 FAILED pluginJSON = ["NfcPlugin521038202","NfcPlugin","makeReadOnly",[]]
2024-07-16 11:01:03.619020+0700 tagReaderSession ended
```