chariotsolutions/phonegap-nfc #410: iOS - Scan and Write

AndreFWeber, 2020-05-18:
Version: phonegap-nfc 1.1.1 "NFC"

Issue:
Whenever I try to use the scanDef() or scanNdef() I am receiving an error message as follows:
"Session invalidated by user"

On writing I receive an error as follow:
"Invalid NDEF Message"

This is my Ionic code:
```
const login: string = "hello";
const message = window.ndef.textRecord(login, null, null);
nfc.write(teste, (data) => {
   console.log('write ok', data);
}, (err) => {
   console.log('write FAIL', err);
});
```

AndreFWeber, 2020-05-18:
I am not a objective C programmer, but I have done some debugging on the write issue to try to solve it, and I've noticed that I can't get through the for loop of writeTag method on NfcPlugin. The code breaks on the following line:

```NSNumber *tnfNumber = [recordData objectForKey:@"tnf"];```
Error: objectForKey unrecognized selector sent to instance

I have also noticed that recordData is actually returning the [command argumentAtIndex:0] keys already. So, in this case recordData hold the value of @"tnf". 
In order to test, I removed the for loop and replaced that line and the following 3 lines that use the recordDate:

```NSNumber *tnfNumber = [[command argumentAtIndex:0] objectForKey:@"tnf"];```

It has passed, but now I am stuck at this line, as the record is always nil:

```            NFCNDEFPayload *record = [[NFCNDEFPayload alloc] initWithFormat:tnf type:type identifier:identifier payload:payload];```

Again, I am not an objective C programmer, so my analysis could be all wrong. But this might be the root of my problem. 

But anyways, is there any configurations or updates I might be missing? 
I am using Version 11.2.1 (11B500) and testing on an iPhone XR, iOS 13.

mbakker96, 2020-06-01:
I've almost the same problem. When running the `scanTag` or `scanNdef` function it returns in the Xcode terminal te following response:

I'm building with the newest Ionic & capacitor. 

```
To Native Cordova ->  NfcPlugin scanNdef NfcPlugin964517628 ["options": []]
2020-06-01 14:25:11.900866+0200 App[1012:186004] scanNdef
2020-06-01 14:25:11.902062+0200 App[1012:186004] shouldUseTagReaderSession 0
2020-06-01 14:25:11.902173+0200 App[1012:186004] callbackOnSessionStart 0
2020-06-01 14:25:11.902218+0200 App[1012:186004] returnTagInCallback 1
2020-06-01 14:25:11.902258+0200 App[1012:186004] returnTagInEvent 0
2020-06-01 14:25:11.902903+0200 App[1012:186004] Using NFCNDEFReaderSession
2020-06-01 14:25:11.903737+0200 App[1012:186222] [CoreNFC] 00000002 80de1300 -[NFCNDEFReaderSession beginSessionWithConfig:]:324  error:Error Domain=NFCError Code=202 "Session invalidated unexpectedly" UserInfo={NSLocalizedDescription=Session invalidated unexpectedly}, errorCode: 0xca
2020-06-01 14:25:11.903839+0200 App[1012:186224] readerSession ended
2020-06-01 14:25:11.903921+0200 App[1012:186224] sendError: Session invalidated unexpectedly
⚡️  [log] - Session invalidated unexpectedly
```

don, 2020-06-18:
@AndreFWeber the message passed to write needs to be an array of messages. Your example code was passing in something else. Try this code to see if you can successfully write https://github.com/chariotsolutions/phonegap-nfc/issues/407#issuecomment-646282570 then update your typescript example.

don, 2020-06-18:
@mbakker96 I don't think that's the same problem.

don, 2020-07-09:
The message passed to write must be an **array** of NDEF Records. Even if it's just one message, create an array.

    var message = [
        ndef.textRecord("Hello phonegap-nfc")
    ];

    nfc.write(
        message,
        success => console.log("Wrote Message"),
        error => console.log(error)
    );

closed 2020-07-09 by don